isaac-sim/IsaacSim #5: Working CUDA ARM

johnnynunez, 2025-06-14:
I'm porting isaac sim and isaac lab to CUDA ARM for cuda-tegra and sbsa.
I will make some PR's

vick-yu, 2025-06-14:
Thank you for your interest in porting Isaac Sim and Isaac Lab to CUDA ARM for cuda-tegra and SBSA platforms.

At this time, the Isaac Sim repository is not yet open for external code contributions or pull requests. We appreciate your enthusiasm and encourage you to continue your development work. When the repository opens for community contributions, we will announce it in the project discussions and update the contribution guidelines accordingly.

For now, if you would like to discuss your progress or share your findings, please feel free to continue the conversation in this issue.

johnnynunez, 2025-06-14:
> Thank you for your interest in porting Isaac Sim and Isaac Lab to CUDA ARM for cuda-tegra and SBSA platforms.
>
> At this time, the Isaac Sim repository is not yet open for external code contributions [...]

Yes, i will use this thread to communicate my news. I did some progress

johnnynunez, 2025-06-15:
i can import isaac sim on aarch64 but some packages are missing for aarch64 and are internally by nvidia

hmazhar-nv, 2025-06-15:
ARM support for isaac sim is in progress, no ETA yet, we are still working on adding ARM support for upstream dependencies. 
Python 3.12 support is expected in the next major kit-kernel version

johnnynunez, 2025-06-16:
> ARM support for isaac sim is in progress, no ETA yet, we are still working on adding ARM support for upstream dependencies. Python 3.12 support is expected in the next major kit-kernel version

thank you! If you could prioritize... I have GH200/GB200 and Jetson Orin to test it.
I would do it for myself, (i ported more than 70 libraries fro cuda arm), but... i depends of some private files from nvidia in this case

tekntrash, 2025-07-05:
This lack of support for jetsons may explain why we get when building:

Script dir: /root/IsaacSim
Traceback (most recent call last):
  File "/root/IsaacSim/tools/repoman/repoman.py", line 31, in <module>
    omni.repo.man.main(REPO_ROOT)
  File "/root/IsaacSim/_repo/deps/repo_man/omni/repo/man/entry.py", line 864, in main
    swap_python_executable(repo_config, options, argv_backup)
  File "/root/IsaacSim/_repo/deps/repo_man/omni/repo/man/python_exec.py", line 142, in swap_python_executable
    executable_path = process_python_executable_config(python_config)
  File "/root/IsaacSim/_repo/deps/repo_man/omni/repo/man/python_exec.py", line 43, in process_python_executable_config
    raise Exception(error_message)
Exception: A `python_executable` configuration block was detected, but `python_executable_path` is not set. Check your repo.toml.

johnnynunez, 2025-07-05:
> This lack of support for jetsons may explain why we get when building:
> 
> Script dir: /root/IsaacSim Traceback (most recent call last): File "/root/IsaacSim/tools/repoman/repoman.py", line 31, in omni.repo.man.main(REPO_ROOT) File "/root/IsaacSim/_repo/deps/repo_man/omni/repo/man/entry.py", line 864, in main swap_python_executable(repo_config, options, argv_backup) File "/root/IsaacSim/_repo/deps/repo_man/omni/repo/man/python_exec.py", line 142, in swap_python_executable executable_path = process_python_executable_config(python_config) File "/root/IsaacSim/_repo/deps/repo_man/omni/repo/man/python_exec.py", line 43, in process_python_executable_config raise Exception(error_message) Exception: A `python_executable` configuration block was detected, but `python_executable_path` is not set. Check your repo.toml.

that is not the problem, the problem is that there are some private packages that not distributed for aarch64, so be patient :)

johnnynunez, 2025-07-16:
How is it going?

martincerven, 2025-08-01:
Also release Isaac Sim for Jetsons not only Spark thanks 💚

johnnynunez, 2025-08-11:
Is 5.0 GA working with cuda arm?

vick-yu, 2026-04-19:
Hi @johnnynunez,                                                                                                                                                                                                                                 
                                                                                                                                                                                                                                                    
   Isaac Sim **aarch64 builds have been available since 5.1.0**. See the [5.1.0 downloads](https://docs.isaacsim.omniverse.nvidia.com/5.1.0/installation/download.html) and [6.0.0                                                                  
 downloads](https://docs.isaacsim.omniverse.nvidia.com/6.0.0/installation/download.html).                                                                                                                                                           
                                                                                                                                                                                                                                                    
   Note that the 6.0.0 [system requirements](https://docs.isaacsim.omniverse.nvidia.com/6.0.0/installation/requirements.html) currently list **NVIDIA DGX Spark** as the supported aarch64 platform. Support for other ARM platforms (GH200,        
 Jetson) is not listed at this time.                                                                                                                                                                                                                
                                                                                                                                                                                                                                                    
   Closing as ARM builds are now available. If you have questions about specific platform support, please open a new discussion.

closed 2026-04-19 by vick-yu

johnnynunez, 2026-04-20:
> Hi [@johnnynunez](https://github.com/johnnynunez),
>
> Isaac Sim **aarch64 builds have been available since 5.1.0**. See the [5.1.0 [...]

yeah yeah, Now i am nvidian since 25th august :)