isaac-sim/IsaacSim #6: Missing isaacsim.simulation_app in build.sh

moribots, 2025-06-15:
I am trying to build the `feature/isaacsim_5_0` branch on a new machine with a 5080 GPU, following the repository's instructions.

https://github.com/isaac-sim/IsaacLab/tree/feature/isaacsim_5_0

The `./build.sh` script reports "BUILD SUCCEEDED", but the resulting build is incomplete. The `find _build -name "simulation_app*"` command returns nothing. This causes all downstream steps to fail with a `ModuleNotFoundError: No module named 'isaacsim.simulation_app'.`

System Info:

GPU: RTX 5080
OS: Ubuntu 22.04 (or your version)
I have confirmed all build dependencies appear to be installed and there are no errors in the build log.
It seems the build script is silently failing to create the `isaacsim.simulation_app` extension.

hmazhar-nv, 2025-06-15:
Can you try with -L, by default find does not follow symlinks

`find -L _build -name "simulation_app*"
_build/linux-x86_64/release/exts/isaacsim.simulation_app/isaacsim/simulation_app
_build/linux-x86_64/release/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py`

moribots, 2025-06-16:
Thank you, that command showed the correct files:
```
(base) moribots:~/Projects/IsaacSim: (main)$ find -L _build -name "simulation_app*"
_build/linux-x86_64/release/exts/isaacsim.simulation_app/isaacsim/simulation_app
_build/linux-x86_64/release/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py
(base) moribots:~/Projects/IsaacSim: (main)$ 
```

I will keep digging to figure out why my IsaacLab could not find this extension

moribots, 2025-06-16:
I'll close this and open an IsaacLab ticket

closed 2025-06-16 by moribots