isaac-sim/IsaacSim #11: [Bug Report] AR Extension streams to AVP in VR

julZuern, 2025-06-16:
### Describe the bug
Today we switched from isaac sim 4.5 to 5.0 to stabilize our AR Streaming with kit 107. In general, multiple issues and (as of a 1hr run) memory leaks have been fixed. The only issue is that the streaming to the Apple Vision Pro is always in VR (the AR tab functionally seems to be an exact copy of the VR tab)

### Steps to reproduce 
1. Start the isaac-sim.xr.vr.bat
2. Enable the AVP configuration extension
3. Select the AR tab and start the cloudXR Server
4. Open the TeleOp app (see below) and connect to the Isaac sim host
5. [The scene should be shown with the UI for controlling the teleop... in VR]

### System Info
**Isaac Sim Host**
Commit: https://github.com/isaac-sim/IsaacSim/commit/2c6f5aefdb6322aa44951e71fcfb973ad413c774
Isaac Sim Version: 5.0.0 (main branch)
OS: Win 11
GPU: RTX 4000 ADA SFF
CUDA: 12.4
GPU Driver: 553.24

**AVP**
VisionOS 2.5 
NVIDIA TeleOp App for IsaacLab https://github.com/[isaac-sim/isaac-xr-teleop-sample-client-apple 
(Both with the default xr framework 1.0.0 and with an updated dependency https://github.com/NVIDIA/cloudxr-framework/tree/release/5.0)

### Additional context
Tested the same setup under 4.5 with the cloudXR41 Server in the AR tab. The Connection was quite unreliable but it was actual AR.

rthaker01, 2025-06-16:
Hi @julZuern  I believe you're using incorrect extension. You should use Teleop using this docs in Isaac Lab: *https://isaac-sim.github.io/IsaacLab/main/source/how-to/cloudxr_teleoperation.html*

julZuern, 2025-06-17:
> Hi [@julZuern](https://github.com/julZuern) I believe you're using incorrect extension, in order to use AR, you need to use AR extension an not to use VR app that you're trying to do. You should use Teleop using this docs in Isaac Lab: _https://isaac-sim.github.io/IsaacLab/main/source/how-to/cloudxr_teleoperation.html_

Hey, I think I need some more clarification on your comment. I did use the AR Extension from within the xr-configured isaac sim launch script. That worked previously and to my knowledge is the intended workflow here. I never stumbled upon a dedicated VR app. The TeleOp app you are referring to appears to be the same as mentioned in my issue. 

For clarification: I do not intend to use the TeleOp functionalities of the AVP app. I use the app as a starting point to use the CloudXR Framework for streaming my viewport. (Eventually I plan to add some interaction with the viewport)

julZuern, 2025-06-17:
Additional info regarding a setup with CUDA 12.8, Driver 573.24 and 2 RTX 4000 ADA (non SFF):
- Isaac Sim crashes upon connecting the AVP, deactivating one graphics card fixes _this_ issue.
- With one RTX 4000 ADA the behavior was the same as described above.