isaac-sim/IsaacSim #12: Observed flickering with primitive mesh geometry on RTX-4090 machine

brianmuldoon27, 2025-06-17:
Hello team, 

I have observed some flickering on the surfaces of primitive mesh geometries when loading and running with a basic IsaacSim scene. I see this issue on a RTX-4090 machine. The brief video below illustrates the issue, with steps to reproduce further below.

https://github.com/user-attachments/assets/cd4c5c6a-8d90-47cd-87db-99e408c5aa4e

### Steps to Reproduce

1. Install Isaac Sim 5.0.0 per the installation instructions in the [README](https://github.com/isaac-sim/IsaacSim/blob/main/README.md)
2. Launch Isaac Sim with 
```
cd _build/linux-x86_64/release
./isaac-sim.sh
```
3. Wait for `Isaac Sim Full App is loaded.` terminal printout from `./isaac-sim.sh`
4. Spawn a `Flat Grid` ground plane in the GUI by clicking `Create > Environments > Flat Grid`
5. Spawn primitives `Cube` and `Cylinder` meshes in GUI by clicking `Create > Mesh > Cube/Cylinder`
6. For the newly added `Cube` and `Cylinder` meshes, apply `Rigid Body with Colliders Preset` properties
7. Step the simulation by clicking the play button in the GUI
8. Observe flickering on the surfaces of the primitive meshes 

### Hardware
Here are the drivers I am using on my RTX4090 workstation,
```  
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.144.03             Driver Version: 550.144.03     CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4090        On  |   00000000:01:00.0  On |                  Off |
| 30%   45C    P2            193W /  480W |    5443MiB /  24564MiB |     55%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
```

Thank you for your help!

Genozen, 2025-06-17:
I'm on a RTX4090, and was able to replicate this issue.

brianmuldoon27, 2025-06-17:
Thanks for reproducing this @Genozen! This is an interesting result. I will leave the issue as open for now since this might be a symptom of a larger issue

vick-yu, 2025-06-30:
Please install the latest production branch version driver ([570.169](https://www.nvidia.com/en-us/drivers/details/247720/)) and check if you still have the issue.

vick-yu, 2025-07-02:
We were able to reproduce the issue using the repository. However, we could not reproduce the same issue with our latest internal version of Isaac Sim. This suggests that the problem may have been addressed in next repo update.

vick-yu, 2025-08-10:
We have not received a response regarding the verification results with the latest update. Therefore, we will be closing this issue. If the problem persists, please feel free to open a new issue.

closed 2025-08-10 by vick-yu