isaac-sim/IsaacSim #16: Not able to compile under Windows 11

sn3ek, 2025-06-17:
I am trying to compile Isaac Sim 5.0 on Windows 11. I have VSCode 2022 and the latest Windows SDK installed but I receive the following trace:

```
[2025-06-16 23:28:28,999][ERROR][omni.repo.man.utils.run_process_return_output:360] error running: ['C:/Users/shared-omniverse/omniverse/IsaacSim/_build/host-deps\\msvc\\MSBuild\\Current\\Bin\\MSBuild.exe', '/NoWarn:MSB8003,MSB8005', 'C:/Users/shared-omniverse/omniverse/IsaacSim/_compiler\\vs2022\\isaac-sim.sln', '/p:Configuration=release;Platform=x64', '/verbosity:normal', '/m:6'], code: 1, message: ""
[2025-06-16 23:28:29,763][WARNING][omni.repo.man.version.get_tool_version:312] Retrieving repo_build version via VERSION file.
[2025-06-16 23:28:29,772][ERROR][omni.repo.man.exceptions_utils.print_log:182] BuildError exception stack trace dumped to logfile C:/Users/shared-omniverse/omniverse/IsaacSim/_repo/repo.log. Stacks dumped to console via `--verbose` or `--tracebacks`. 
BuildError: BUILD FAILED for release
[2025-06-16 23:28:29,772][ERROR][omni.repo.emit_stack_to_logfile:156] BuildError encountered while running repo_build via ['C:\\Users\\shared-omniverse\\omniverse\\IsaacSim\\tools\\repoman\\repoman.py', 'build', '--verbose']:
Traceback (most recent call last):
  File "C:\Users\shared-omniverse\omniverse\IsaacSim\_repo\deps\repo_man\omni\repo\man\entry.py", line 138, in run
    options.func(options)
  File "C:\Users\shared-omniverse\omniverse\IsaacSim\_repo\deps\repo_man\omni\repo\man\entry.py", line 90, in <lambda>
    func=lambda args, func=func, config=self.merged_tool_config: func(args, config),
                                                                 ^^^^^^^^^^^^^^^^^^
  File "C:\Users\shared-omniverse\omniverse\IsaacSim\_repo\deps\repo_build\omni\repo\build\main.py", line 1576, in run_repo_tool
    run_build(options, repo_folders, None, config)
  File "C:\Users\shared-omniverse\omniverse\IsaacSim\_repo\deps\repo_build\omni\repo\build\utils.py", line 49, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\shared-omniverse\omniverse\IsaacSim\_repo\deps\repo_build\omni\repo\build\main.py", line 1229, in run_build
    build_fn(
  File "C:\Users\shared-omniverse\omniverse\IsaacSim\_repo\deps\repo_build\omni\repo\build\main.py", line 738, in build
    build_with_msbuild(
  File "C:\Users\shared-omniverse\omniverse\IsaacSim\_repo\deps\repo_man\omni\repo\man\utils.py", line 1218, in wrapper
    result = func(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\shared-omniverse\omniverse\IsaacSim\_repo\deps\repo_build\omni\repo\build\build.py", line 539, in build_with_msbuild
    do_build(args, config, error_processing, teamcity_block_name=f"Windows Build {config.capitalize()}")
  File "C:\Users\shared-omniverse\omniverse\IsaacSim\_repo\deps\repo_man\omni\repo\man\nvteamcity.py", line 177, in wrapper
    result = func(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\shared-omniverse\omniverse\IsaacSim\_repo\deps\repo_build\omni\repo\build\build.py", line 333, in do_build
    raise BuildError(f"BUILD FAILED for {config}")
omni.repo.build.exceptions.BuildError: BUILD FAILED for release
[2025-06-16 23:28:29,897][ERROR][omni.repo.man.utils.run_process:264] error running: _repo\python\python.exe C:\Users\shared-omniverse\omniverse\IsaacSim\tools\repoman\repoman.py build --verbose, code: 1, message: ""
```

I tried it multiple times but it was not successful yet.

vick-yu, 2025-08-09:
Since there has been no recent activity on this issue, I will be closing it. If you are still having this problem, you can open a new issue with a reference to this one and include any latest details.

zilingzhang, 2025-08-09:
I found "vcpkg integrate install" caused my problem. 
Removing it with "vcpkg integrate remove" fixed the build with tbb split issue.