isaac-sim/IsaacSim #17: [Bug]: unable to install/load the ROS2 bridge extension

lesurJ, 2025-06-17:
Hello,

Thank you for making the code open source 😃!

I have one bug with the ROS2 Bridge extension. When trying to install it from the extension manager, I have the `Could not import system rclpy: No module named 'rclpy._rclpy_pybind11'` error, despite this library being installed on the system.

```
[488.194s] [ext: isaacsim.ros2.bridge-4.8.3] startup
[488.206s] Attempting to load system rclpy
[488.208s] Could not import system rclpy: No module named 'rclpy._rclpy_pybind11'
The C extension '/opt/ros/jazzy/lib/python3.12/site-packages/_rclpy_pybind11.cpython-311-x86_64-linux-gnu.so' isn't present on the system. Please refer to 'https://docs.ros.org/en/jazzy/How-To-Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system' for possible solutions
[488.208s] Attempting to load internal rclpy for ROS Distro: jazzy
[488.210s] Could not import internal rclpy: No module named 'rclpy._rclpy_pybind11'
The C extension '/opt/ros/jazzy/lib/python3.12/site-packages/_rclpy_pybind11.cpython-311-x86_64-linux-gnu.so' isn't present on the system. Please refer to 'https://docs.ros.org/en/jazzy/How-To-Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system' for possible solutions
[488.210s] To use the internal libraries included with the extension please set the following environment variables to use with FastDDS (default) or CycloneDDS (ROS2 Humble only) before starting Isaac Sim:

FastDDS (default):
export ROS_DISTRO=jazzy
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/jazzy/lib

OR

CycloneDDS (ROS2 Humble only):
export ROS_DISTRO=jazzy
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/jazzy/lib

```

I can confirm that my ROS2 Jazzy install works fine as I am able to use it.
Moreover, the `_rclpy_pybind11.cpython-311-x86_64-linux-gnu.so` library exists but is installed under `/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/` and NOT under `/opt/ros/jazzy/lib/python3.12/site-packages/` as used by the extension. Hence the import error.

Also, i confirm having exported the environment variables for CycloneDDS as indicated.

Isaac Sim Version: 5.0
Operating System: Ubuntu 24.04
GPU Information:GeForce RTX4080, with driver 535.230.02
ROS2: Jazzy, installed from binaries

PS: Also confusing to see the CycloneDDS indicating Humble only but exporting the ROS_DISTRO as jazzy.

blooop, 2025-06-17:
I also get this error on my system.

```
Isaac Sim Version: 5.0
Operating System: Ubuntu 24.04
GPU Information:GeForce RTX4090M, with driver 535.230.02
ROS2: Jazzy, installed from binaries
```

parzival2, 2025-06-17:
Also have the same problem 
```
Isaac Sim Version: 5.0
Operating System: Ubuntu 22.04
ros2 version: humble
```
Getting an error like this when running Isaac-Sim
```
[11.669s] Attempting to load system rclpy
[11.693s] Could not import system rclpy: No module named 'rclpy._rclpy_pybind11'
The C extension '/opt/ros/humble/lib/python3.10/site-packages/_rclpy_pybind11.cpython-311-x86_64-linux-gnu.so' isn't present on the system. Please refer to 'https://docs.ros.org/en/humble/Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system' for possible solutions
[11.693s] Attempting to load internal rclpy for ROS Distro: humble
[11.695s] Could not import internal rclpy: No module named 'rclpy._rclpy_pybind11'
The C extension '/opt/ros/humble/lib/python3.10/site-packages/_rclpy_pybind11.cpython-311-x86_64-linux-gnu.so' isn't present on the system. Please refer to 'https://docs.ros.org/en/humble/Guides/Installation-Troubleshooting.html#import-failing-without-library-present-on-the-system' for possible solutions
[11.695s] To use the internal libraries included with the extension please set the following environment variables to use with FastDDS (default) or CycloneDDS (ROS2 Humble only) before starting Isaac Sim:

FastDDS (default):
export ROS_DISTRO=humble
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/humble/lib

OR

CycloneDDS (ROS2 Humble only):
export ROS_DISTRO=humble
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/humble/lib
```
The thing is instead of this `/opt/ros/humble/lib/python3.10/site-packages/_rclpy_pybind11.cpython-311-x86_64-linux-gnu.so` I have `/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/_rclpy_pybind11.cpython-310-x86_64-linux-gnu.so`, so it looks like there is a version mismatch. Is it coming from `ros2-bridge` extension?

k3street, 2025-06-18:
Same issue 24.04, ros version Jazzy which says it is supported

sattwik-sahu, 2025-10-07:
Is there a way to use internal ROS2 with the `pip` installed version of isaacsim?

Toni-SM, 2025-10-07:
Hi @sattwik-sahu 

> Is there a way to use internal ROS2 with the pip installed version of isaacsim?

Yes, just `export` the variables shown in the terminal when launching `isaacsim`.

For, example (on my maching)
```
...
[13.079s] To use the internal libraries included with the extension please set the following environment variables to use with FastDDS (default) or CycloneDDS before starting Isaac Sim:

FastDDS (default):
export ROS_DISTRO=humble
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/env_isaacsim/lib/python3.11/site-packages/isaacsim/exts/isaacsim.ros2.bridge/humble/lib

```