isaac-sim/IsaacSim #19: Extension Test Failed: isaacsim.core.api

singh-jayant, 2025-06-18:
## My Setup
- RTX 5070 Ti
- Ubuntu 24.04
- Driver 575

## Problem
The `run_all_tests.sh` script seems to be failing and I am not sure if this is going to cause problems or not. The simulator runs and is able to load USDs.

```log
[fail] Extension test failed.

[fail] Extension Test failed. Details:
    Cmdline: ./../kit/kit /home/jayant/isaacsim/extscache/omni.kit.test-2.0.1+b0a86421.lx64.r.cp311/apps/omni.app.test_ext.kit --enable isaacsim.core.api-4.6.1 --/log/flushStandardStreamOutput=1 --/app/name=exttest_isaacsim_core_api --/log/file='/home/jayant/.local/share/ov/data/_testoutput/exttest_isaacsim_core_api/exttest_isaacsim_core_api_2025-06-18T09-07-42_0.log' --/exts/omni.kit.test/testOutputPath='/home/jayant/.local/share/ov/data/_testoutput/exttest_isaacsim_core_api' --/exts/omni.kit.test/extTestId='isaacsim.core.api' --/crashreporter/dumpDir='/home/jayant/.local/share/ov/data/_testoutput/exttest_isaacsim_core_api' --/crashreporter/preserveDump=1 --/crashreporter/gatherUserStory=0 --/app/isTestRun=1 --/rtx-transient/dlssg/enabled=false --ext-folder /home/jayant/isaacsim/kit/extsPhysics --ext-folder ../exts --ext-folder ../extscache --ext-folder ../extsDeprecated --ext-folder ../apps --enable omni.kit.test --/exts/omni.kit.test/runTestsAndQuit=true --/exts/omni.kit.test/includeTests/0='isaacsim.core.api.*' --/exts/omni.kit.test/includeTests/1='isaacsim.core.api.tests.*' --allow-root --/telemetry/mode=test --/crashreporter/data/testName=ext-test-isaacsim.core.api --no-window --/exts/omni.kit.test/pyCoverageEnabled=False --enable isaacsim.core.cloner --enable omni.kit.renderer.core --no-window --allow-root --/app/asyncRendering=0 --/app/asyncRenderingLowLatency=0 --/app/fastShutdown=1 --/app/file/ignoreUnsavedOnExit=1 --/app/hydraEngine/waitIdle=0 --/app/renderer/skipWhileMinimized=0 --/app/renderer/sleepMsOnFocus=0 --/app/renderer/sleepMsOutOfFocus=0 --/app/settings/fabricDefaultStageFrameHistoryCount=3 --/app/settings/persistent=0 --/app/viewport/createCameraModelRep=0 --/crashreporter/skipOldDumpUpload=1 --/exts/omni.usd/locking/onClose=0 --/omni/kit/plugin/syncUsdLoads=1 --/omni/replicator/asyncRendering=0 --/persistent/app/stage/upAxis="Z" --/persistent/app/viewport/defaults/tickRate=120 --/persistent/app/viewport/displayOptions=31951 --/persistent/omni/replicator/captureOnPlay=1 --/persistent/omnigraph/updateToUsd=0 --/persistent/physics/visualizationDisplayJoints=0 --/persistent/renderer/startupMessageDisplayed=1 --/persistent/simulation/defaultMetersPerUnit=1.0 --/persistent/simulation/minFrameRate=15 --/renderer/multiGpu/autoEnable=0 --/renderer/multiGpu/enabled=0 --/rtx-transient/dlssg/enabled=0 --/'rtx-transient'/resourcemanager/enableTextureStreaming=1 --/rtx/descriptorSets=360000 --/rtx/hydra/enableSemanticSchema=1 --/rtx/hydra/materialSyncLoads=1 --/rtx/materialDb/syncLoads=1 --/rtx/newDenoiser/enabled=1 --/rtx/reservedDescriptors=900000 --vulkan --/app/useFabricSceneDelegate=true --/app/player/useFixedTimeStepping=false --/app/runLoops/main/rateLimitEnabled=false --/app/runLoops/main/manualModeEnabled=true
    Cmdline to run the extension tests: ./../kit/tests-isaacsim.core.api.sh
    Return code: 0 (0x00000000)
    Failure reason(s): 
        Matched 2 fail pattern '*[error]*' in stdout: 
            '2025-06-18T07:08:30Z [45,862ms] [Error] [omni.physx.plugin] PhysX error: PxArticulationLink::setGlobalPose(): it is illegal to call this method if PxSceneFlag::eENABLE_DIRECT_GPU_API is enabled!, FILE /builds/omniverse/physics/physx/source/physx/src/NpArticulationLink.cpp, LINE 353'
            '2025-06-18T07:08:30Z [45,862ms] [Error] [omni.physx.plugin] PhysX error: PxArticulationLink::setGlobalPose(): it is illegal to call this method if PxSceneFlag::eENABLE_DIRECT_GPU_API is enabled!, FILE /builds/omniverse/physics/physx/source/physx/src/NpArticulationLink.cpp, LINE 353'
    Details:

||||||||||||||||||||||||||||||  [EXTENSION TEST FAILED: isaacsim.core.api]  ||||||||||||||||||||||||||||||

```

hmazhar-nv, 2025-06-19:
This is a known issue, we are working on a fix internally. You should not see issues during normal use as a result of this

vick-yu, 2025-07-23:
Please note that run_all_tests.sh has been removed in the latest update. Users can now run tests individually from the tests folder.

closed 2025-07-23 by vick-yu