isaac-sim/IsaacSim #26: [BUG]: Hand tracking doesn't seem to work in IsaacSim on windows

kenchen3000, 2025-06-19:
### Description

Hi,
I am trying hand tracking example with IsaacSim on Windows, with OpenXR and Quest 3. But there seems to be problem and no hand tracking message is detected.

Under AR panel -> XCR: Capture and Replay, it shows error:
Capture layer is not present in windows registry, refer to README.md to install it.

In command prompt window, it also shows:
[BodyApiPlugin] [WARN]: RuntimeInputInterface V6 not found

Could help to check how to make it work? thank you very much!

The examples i tried:
`.\isaac-sim.xr.vr.bat source\standalone_examples\api\isaacsim.xr.openxr\hand_tracking\hand_tracking_sample.py`

`./isaaclab.bat -p scripts/tools/record_demos.py --task Isaac-Stack-Cube-Franka-IK-Rel-v0 --teleop_device handtracking_abs --dataset_file ./datasets/dataset.hdf5 --num_demos 10`

None of these 2 works. 

### Isaac Sim version

4.5.0, 5.0.0

### Operating System (OS)

windows 11

### GPU Name

RTX A6000

### GPU Driver and CUDA versions

573 and Cuda 12

### Logs

2025-06-19 18:08:19 [511,136ms] [Warning] [omni.fabric.plugin] removePath called on non-existent path /_xr/stage/xrAnchor

2025-06-19 18:08:19 [511,142ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling setLocalMatrix for prim /_xr/stage/xrAnchor that has not been populated
19/06 11:09:03.804 {DEBUG}   [OXR] xrNegotiateLoaderRuntimeInterface loading 1.1.46
19/06 11:09:04.546 {DEBUG}   [OXR] xrNegotiateLoaderRuntimeInterface loading 1.1.46
[BodyApiPlugin][ INFO]: BodyApiClientInterface7 initialized.
[BodyApiPlugin][ INFO]: BodyModelParamsClientInterface initialized.
[BodyApiPlugin][ INFO]: BeginSession called for BodyApi Plugin
[BodyApiPlugin][ WARN]: RuntimeInputInterface V6 not found
[BodyApiPlugin][ INFO]: EndSession called for BodyApi Plugin

zwdoescode, 2025-07-10:
That example illustrates the interaction between our IsaacXRTeleopClient application and for that reason, it will not display the hands without a specific message from the client.

You should be able to work around this by changing [tracking_enabled](https://github.com/isaac-sim/IsaacSim/blob/main/source/standalone_examples/api/isaacsim.xr.openxr/hand_tracking/hand_tracking_sample.py#L31C1-L31C17) to true.

The message expected by the app is tied to a specific client UI element to start/stop a teleop session from the headset. If your client does not have this message implemented, it will not display the hand tracking data as the application defaults to the “stoped” state. A workaround is to modify the sample application [hand_tracking_sample.py](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fisaac-sim%2FIsaacSim%2Fblob%2Fmain%2Fsource%2Fstandalone_examples%2Fapi%2Fisaacsim.xr.openxr%2Fhand_tracking%2Fhand_tracking_sample.py%23L40&data=05%7C02%7Crwiltz%40nvidia.com%7Cc1f2ecb42a6d4382087008ddbf3cebe9%7C43083d15727340c1b7db39efd9ccc17a%7C0%7C0%7C638876990628970488%7CUnknown%7CTWFpbGZsb3d8eyJFbXB0eU1hcGkiOnRydWUsIlYiOiIwLjAuMDAwMCIsIlAiOiJXaW4zMiIsIkFOIjoiTWFpbCIsIldUIjoyfQ%3D%3D%7C0%7C%7C%7C&sdata=WTSUudqsZSWeT3apMilVXdfqMCwyBMan8QnzNJtSh9s%3D&reserved=0) so that this message is not required and it starts in a “start” state.

zwdoescode, 2025-08-07:
I will close this since this issue has been idle. But please feel free to re-open it if the issue persists

closed 2025-08-07 by zwdoescode