isaac-sim/IsaacSim #32: [BUG]: Debug Print Node does not work with Impulse or one off events

ume-daniel, 2025-06-25:
### Description

The debug print node does not print to screen when you connected to an impulse event or a stage event or any event that fires once. It does work with an onTick or if you button mash the impulse event button. Works in 4.5 and kit 107.3 apps.

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 22.04

### GPU Name

A10

### GPU Driver and CUDA versions

550.54.17

vick-yu, 2025-07-03:
> The debug print node does not print to screen when you connected to an impulse event or a stage event or any event that fires once. It does work with an onTick or if you button mash the impulse event button. Works in 4.5 and kit 107.3 apps.
Did you mean that it doesn’t work on 4.5 with the default Kit version 106.5.0? If possible, could you share a video clip demonstrating how it works on 4.5 and on 107.3?

ume-daniel, 2025-07-03:
It does not work in Isaac Sim 5.0.

It does work everywhere else - Kit 107.3, 106.5 and Isaac 4.5.

Something about 5.0 is clearing the viewport display of the debug print instead of letting it be shown for the usual 3 seconds. I believe that's why using onTick is working because it constantly drawing it. It is not isolated to Impulse events, any event that only fires once shows this issue.

vick-yu, 2025-07-03:
This USD doesn't work on my side even on 4.5. Could you share how you trigger the on impulse event? 
Are you able to see the print in the code on 5.0?

ume-daniel, 2025-07-04:
Here is a link to a video showing the issue: https://youtu.be/S3Edfe1i2GA

1. The debug print will print to console, but setting it to print to screen does not work when triggered by an impulse.
2. When using a tick, the debug print will print to screen.

vick-yu, 2025-10-15:
This issue has been addressed in Isaac Sim 5.1.0, so I'm closing this GitHub issue.