isaac-sim/IsaacSim #36: [BUG]: Isaac Sim 5.0 Crashes on Launch

Jack-mir, 2025-06-26:
### Description

I’ve installed Isaac Sim 5.0, but after launching it, the application encounters some errors and then crashes.

### Isaac Sim version

5.0

### Operating System (OS)

window11 pro

### GPU Name

L40S-16Q

### GPU Driver and CUDA versions

Driver:552.74, CUDA : 12.4 

### Logs

[crash] Crash detected in pid 12708 thread 13672
[crash]     Note: some long metadata and file name values below may be truncated.  The
[crash]           full values of each piece of metadata and each full file name can still
[crash]           be found in the file 'C:/Users/ssjackqwe/AppData/Local/ov/data/Kit/Isaac-Sim Full/5.0\895a39e1-9af0-4385-8cac-d16bf550799f.dmp.toml'.
[crash]           This file can be found in either the crash report's '.dmp.zip' file or
[crash]           in the same directory as the crash report file(s):
[crash]             'C:/Users/ssjackqwe/AppData/Local/ov/data/Kit/Isaac-Sim Full/5.0'.
[crash]
[crash] Crash metadata:
[crash]  BuildGitlabJobID = '173804052'
[crash]  CarbSdkVersion = '206.1+release.8995.85e27be5.gl'
[crash]  CrashTime = 'Thu Jun 26 00:52:08 2025 GMT'
[crash]  DriverShaderCacheWrapper = 'disabled'
[crash]  DumpId = '895a39e1-9af0-4385-8cac-d16bf550799f'
[crash]  LastUploadStatus = '0'
[crash]  ProductName = 'OmniverseKit'
[crash]  PythonTracebackStatus = ''py-spy.exe' successfully wrote info to C:/Users/ssjackqwe/AppData/Local/ov/data/Kit/Isaac-Sim Full/5.0/895a39e1-9af0-4385-8cac-d16bf550799f.py.txt'
[crash]  RetryCount = '0'
[crash]  StartupTime = '1750898994'
[crash]  UploadSuccessful = '0'
[crash]  UptimeSeconds = '134'
[crash]  UserStory = ''
[crash]  UserStoryStatus = ''crashreport.gui.exe' exited with code 3'
[crash]  Version = '107.3.0+feature.199947.b0a86421.gl'
[crash]  aftermath_enabled = '1'
[crash]  aftermath_flags = '3'
[crash]  aftermath_status = 'auto-enabled'
[crash]  aftermath_version = '2024.1'
[crash]  appName = 'Isaac-Sim Full'
[crash]  appState = 'started'
[crash]  appVersion = '5.0.0'
[crash]  autoloadExts = ''
[crash]  buildBranch = 'feature'
[crash]  buildCi = 'gl'
[crash]  buildConfig = 'release'
[crash]  buildGitlabJobName = 'kit-build-release-windows-x86_64'
[crash]  buildGitlabJobStage = 'kit-build'
[crash]  buildGitlabPipelineID = '29345780'
[crash]  buildHash = 'b0a86421'
[crash]  buildMajor = '107'
[crash]  buildMinor = '3'
[crash]  buildNumber = '199947'
[crash]  buildPatch = '0'
[crash]  buildVersion = '107.3.0'
[crash]  carboniteFrameworkVersion = '206.1+release.8995.85e27be5.gl'
[crash]  carboniteSdkVersion = '206.1+release.8995.85e27be5.gl'
[crash]  commandLine = '"C:\Users\<username>\isaacsim\_build\windows-x86_64\release\kit\kit.exe"  "C:\Users\<username>\isaacsim\_build\windows-x86_64\release\apps/isaacsim.exp.full.kit"  '
[crash]  cpuCoreLimited = '-1'
[... 115 lines cut ...]
[crash]  workingDirectory = 'C:\Users\<username>\isaacsim\_build\windows-x86_64\release'
[crash]
[crash] Crash files:
[crash]  metadataFile = 'C:/Users/ssjackqwe/AppData/Local/ov/data/Kit/Isaac-Sim Full/5.0\895a39e1-9af0-4385-8cac-d16bf550799f.dmp.toml'
[crash]  upload_file_minidump = 'C:/Users/ssjackqwe/AppData/Local/ov/data/Kit/Isaac-Sim Full/5.0\895a39e1-9af0-4385-8cac-d16bf550799f.dmp'
[crash]  upload_file_traceback = 'C:/Users/ssjackqwe/AppData/Local/ov/data/Kit/Isaac-Sim Full/5.0/895a39e1-9af0-4385-8cac-d16bf550799f.py.txt'
[crash]
[crash] Thread 13672 backtrace follows:
000: rtx.mdltranslator.plugin.dll!carbOnPluginStartup+0x3116
001: rtx.mdltranslator.plugin.dll!carbOnPluginStartup+0x1ee
002: carb.dll!carbReallocate+0x4ed3
003: carb.dll!releaseFrameworkAndTerminate+0xfa7e
004: carb.dll!carbGetErrorApi+0xf447
005: carb.scenerenderer-rtx.plugin.dll!carbOnPluginShutdown+0x9ca50
006: carb.scenerenderer-rtx.plugin.dll!carbOnPluginShutdown+0x9b6fa
007: carb.scenerenderer-rtx.plugin.dll!carbOnPluginShutdown+0xbfb92
008: omni.hydra.rtx.plugin.dll!+0x553c
009: omni.usd.dll!omni::usd::UsdManager::getFoundationPlugins+0xe89
010: omni.usd.dll!gpu::realm::TypeSerializer<omni::usd::hydra::HydraEngineCreationConfig,void>::write+0x74c3
011: omni.usd.dll!gpu::realm::TypeSerializer<omni::usd::hydra::HydraEngineCreationConfig,void>::write+0x770e
012: carb.tasking.plugin.dll!+0x234ab
013: carb.tasking.plugin.dll!+0x18b07
014: carb.tasking.plugin.dll!+0x18b29
015: KERNELBASE.dll!SetSecurityDescriptorControl+0x90
016: ntdll.dll!RtlUserFiberStart+0x19

vick-yu, 2025-07-02:
Please ensure that your GPU driver version meets the requirement stated in https://docs.omniverse.nvidia.com/dev-guide/latest/common/technical-requirements.html#blackwell.

Jack-mir, 2025-07-04:
> Please ensure that your GPU driver version meets the requirement stated in https://docs.omniverse.nvidia.com/dev-guide/latest/common/technical-requirements.html#blackwell.

I tried Data Center 573.39 576.57 driver in the link you gave me,after extracting the installer, the installer told me that the graphics driver is not compatible with my device,I've check these two driver's support device,it should be fine with L40s the pictures below are my graphics card and the installation failure warning

[image]
[image]
[image]

vick-yu, 2025-07-05:
Thank you for providing the details about your setup. To better assist with troubleshooting, could you please clarify a few points about your environment?

- Are you running Isaac Sim 5.0 on a local workstation, or is this on a virtual machine or cloud instance?
- If this is a cloud or virtualized setup, could you specify which cloud provider or virtualization platform you are using (e.g., AWS, Azure, VMware, etc.)?
- For the L40S-16Q GPU profile, are you using NVIDIA vGPU technology, and if so, is the VM configured with the 16Q profile (16 GB VRAM, full graphics acceleration)?

These details will help us understand your environment and provide more targeted support for the crash you’re experiencing. Thank you!

closed 2025-07-24 by vick-yu