isaac-sim/IsaacSim #39: [BUG]: Failing to load a whole bunch of modules, but app still loads

emirkin, 2025-06-27:
### Description

I am working with a fresh install of Jazzy (according to https://docs.ros.org/en/jazzy/Installation/Ubuntu-Install-Debs.html) and Isaac Sim 5.0 rc3 (commit `cbbda4f`).

Python: 3.13
GCC/G++: 11

After building Isaac Sim (`build.sh`) and launching it according to the instructions in this repo (`.../release/isaac-sim.sh`), I see a lot of failed module imports: `isaacsim.core.simulation_manager`, `isaacsim.core.prims`, `isaacsim.sensors.physics`, `isaacsim.core.cloner`, ..., `isaacsim.sensors.physx.ui`, `isaacsim.ros2.bridge`. Yet, `./isaac-sim.sh` still loads the app.

Should I be concerned about these errors?

### Isaac Sim version

5.0 rc3

### Operating System (OS)

Ubuntu 24.04.2

### GPU Name

RTX 4090

### GPU Driver and CUDA versions

Nvidia 570.169 / CUDA 12.8

Toni-SM, 2025-06-27:
Hi @emirkin 

Could you please share logs files?
See the Common Path Locations in Isaac Sim's [Setup Tips](https://docs.isaacsim.omniverse.nvidia.com/latest/installation/install_faq.html) documentation for details about the locations of the log files.

emirkin, 2025-06-27:
[isaac-sim-modules-not-loading.log](https://github.com/user-attachments/files/20953136/isaac-sim-modules-not-loading.log)

emirkin, 2025-06-27:
If this helps... 
I deleted IsaacSim 5.0 rc3 and installed isaac-sim-standalone@4.5.0-rc.36+release.19112.f59b3005.gl.linux-x86_64.release instead. No error messages, everything loaded fine.

Toni-SM, 2025-06-27:
Hi @emirkin 

According to the logs, problem it/was torch... like reported in https://github.com/isaac-sim/IsaacSim/issues/24.
You either must remove CUDA from your `LD_LIBRARY_PATH` or install CUDA 12.8

```
  File "/home/zhenya/isaacsim/_build/linux-x86_64/release/exts/omni.isaac.ml_archive/pip_prebundle/torch/__init__.py", line 409, in <module>
    from torch._C import *  # noqa: F403
    ^^^^^^^^^^^^^^^^^^^^^^
ImportError: /home/zhenya/isaacsim/_build/linux-x86_64/release/exts/omni.isaac.ml_archive/pip_prebundle/torch/lib/../../nvidia/cusparse/lib/libcusparse.so.12: undefined symbol: __nvJitLinkCreate_12_8, version libnvJitLink.so.12
```

emirkin, 2025-06-27:
@Toni-SM , I do have CUDA 12.8, as stated in issue details.

I suspect the problem lies elsewhere. Specifically, I have conda+python 3.13 active and `build.sh` must have not picked it up, opting to use some internal version (kit?) of python instead (3.10?). That would explain why modules cannot be found.
 I have already replaced 5.0 with 4.5.0, so I don't have a quick way to verify that. Does what I am suggesting seem like it has legs?

vick-yu, 2025-07-22:
@emirkin Please see if the solution discussed in https://github.com/isaac-sim/IsaacSim/issues/81 help with the problem you encountered.