isaac-sim/IsaacSim #44: [BUG]: Cannot create warehouse with curve tool

felixchengxh, 2025-06-28:
### Description

I was following the documentation of the [Warehouse Creator Extension](https://docs.isaacsim.omniverse.nvidia.com/5.0.0/digital_twin/warehouse_logistics/ext_omni_warehouse_creator.html), but encountered an issue when trying to draw a curve. When I attempt to use the Curve tool, nothing happens in the viewport. Upon checking the "Curve" window, I noticed that all control points are initialized at [0, 0, 0], and the following error messages appear in the console:

```
2025-06-28T08:35:59Z [37,813ms] [Warning] [omni.warehouse_creator.warehouse_creator] Import error: cannot unpack non-iterable NoneType object
2025-06-28T08:35:59Z [37,813ms] [Error] [omni.warehouse_creator.warehouse_creator] Traceback (most recent call last):
  File "/home/chengshianhuei/project/isaacsim/_build/linux-x86_64/release/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 220, in _on_curves_event
    self.add_point()
  File "/home/chengshianhuei/project/isaacsim/_build/linux-x86_64/release/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 206, in add_point
    ret = self.add_wall(prev_point, current_point, updated_next_point)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chengshianhuei/project/isaacsim/_build/linux-x86_64/release/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 291, in add_wall
    rot_quat, axis, num_of_wall_segments = self.get_wall_data(current_point, next_point)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: cannot unpack non-iterable NoneType object
```
It seems like the curve is not being correctly initialized based on user input. Could you please advise whether this is a known issue, or if there’s any step I might have missed?
Thank you!

[image]

[image]

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 22.04

### GPU Name

RTX 5080

### GPU Driver and CUDA versions

Driver Version: 570.153.02, CUDA Version: 12.8

zwdoescode, 2025-07-10:
Thanks for bringing this issue up. I am able to replicate the issue on my side. Let me reach out to the correct person

yyjj25, 2025-08-15:
Hi, I’m also encountering this problem. Could you please let me know if there’s any new information or progress? Thank you!

jfelixre, 2025-08-19:
Hi, I have the same problem

BenjaSpain, 2025-12-10:
I face same error using Warehouse Creator extension.
It does not work for me, sometimes it show the yellow line and I manage to draw some sections , but it is only the exception ; most of times I cannot draw a single block

Environment:

- Isaac Sim Version: 5.1
- WebRTC Streaming Client 1.1.5
- OS: Ubuntu 24.04
- Warehouse Creator ext version: v0.4.4

LOG ERROR:

`2025-12-10T13:53:47Z [204,997ms] [Error] [omni.warehouse_creator.warehouse_creator] Traceback (most recent call last):
  File "/isaac-sim/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 220, in _on_curves_event
    self.add_point()
  File "/isaac-sim/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 206, in add_point
    ret = self.add_wall(prev_point, current_point, updated_next_point)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/isaac-sim/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 291, in add_wall
    rot_quat, axis, num_of_wall_segments = self.get_wall_data(current_point, next_point)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: cannot unpack non-iterable NoneType object

2025-12-10T13:53:56Z [214,399ms] [Warning] [omni.warehouse_creator.warehouse_creator] Import error: cannot unpack non-iterable NoneType object
2025-12-10T13:53:56Z [214,400ms] [Error] [omni.warehouse_creator.warehouse_creator] Traceback (most recent call last):
  File "/isaac-sim/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 220, in _on_curves_event
    self.add_point()
  File "/isaac-sim/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 206, in add_point
    ret = self.add_wall(prev_point, current_point, updated_next_point)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/isaac-sim/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 291, in add_wall
    rot_quat, axis, num_of_wall_segments = self.get_wall_data(current_point, next_point)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: cannot unpack non-iterable NoneType object

2025-12-10T13:53:57Z [215,192ms] [Warning] [omni.kit.window.property.templates.simple_property_widget] PrimKindWidget.build_items took 0.530000133003341 seconds
2025-12-10T13:53:59Z [217,160ms] [Warning] [omni.warehouse_creator.warehouse_creator] Import error: cannot unpack non-iterable NoneType object
2025-12-10T13:53:59Z [217,161ms] [Error] [omni.warehouse_creator.warehouse_creator] Traceback (most recent call last):
  File "/isaac-sim/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 220, in _on_curves_event
    self.add_point()
  File "/isaac-sim/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 206, in add_point
    ret = self.add_wall(prev_point, current_point, updated_next_point)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/isaac-sim/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 291, in add_wall
    rot_quat, axis, num_of_wall_segments = self.get_wall_data(current_point, next_point)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: cannot unpack non-iterable NoneType object

2025-12-10T13:54:04Z [221,628ms] [Warning] [omni.warehouse_creator.warehouse_creator] Import error: cannot unpack non-iterable NoneType object
2025-12-10T13:54:04Z [221,629ms] [Error] [omni.warehouse_creator.warehouse_creator] Traceback (most recent call last):
  File "/isaac-sim/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 220, in _on_curves_event
    self.add_point()
  File "/isaac-sim/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 206, in add_point
    ret = self.add_wall(prev_point, current_point, updated_next_point)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/isaac-sim/extscache/omni.warehouse_creator-0.4.4/omni/warehouse_creator/warehouse_creator.py", line 291, in add_wall
    rot_quat, axis, num_of_wall_segments = self.get_wall_data(current_point, next_point)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: cannot unpack non-iterable NoneType object

`