isaac-sim/IsaacSim #52: [BUG]: Mismatched units found on drag and drop, resolved in current authoring layer path: /World/Object

manavkulshrestha, 2025-06-30:
### Description

Custom USD object is not visible. I was previously using 4.5.0, where the object is visible and works properly. I created it using an `.obj` file and the code in `standalone_examples/api/omni.kit.asset_converter/asset_usd_converter.py` from isaacsim-4.5.0. Is there modification that needs to be done to the USD file for compatibility with 5.0.0?

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 24.04

### GPU Name

RTX 5080

### GPU Driver and CUDA versions

Driver 570.133.07, CUDA 12.8

vick-yu, 2025-07-01:
Could you please check if USD files generated using the asset_usd_converter.py script from Isaac Sim 5.0 still exhibit this issue?

vick-yu, 2025-07-23:
Are you seeing these issues (the invisibility unless use_meter_as_world_unit is set to true and the error message) only in Isaac Sim 5.0, but not in Isaac Sim 4.5? Also, do they occur only with this specific .obj file? If possible, could you please share the .obj file so we can try to reproduce the problem on our end?

vick-yu, 2025-07-23:
I wasn’t able to reproduce this error message (“Mismatched units found on drag and drop, resolved in current authoring layer path: /World/Object”) when opening your ef631a2ce94fae3ab8966911a5afa22c.usd file using Isaac Sim 5.0. Has the issue been resolved on your end?