isaac-sim/IsaacSim #55: [BUG]: Orientations for custom USD objects are incorrect

manavkulshrestha, 2025-07-01:
### Description

Modifying/setting the orientation for a custom USD object seems to result in the incorrect pose and even results in changes to the position when it shouldn't. There's also some odd flickering in the position/orientation of the object:

[orientation flickering view](https://youtu.be/jPHUT8mfQfo)

This is in contrast to 4.5.0 where no such issues existed. Also, the (1, 0, 0, 0) orientation that I spawn the objects at causes a different pose in 5.0.0 compared to 4.5.0:

[orientation mismatch images](https://imgur.com/a/vOocgoy)

To reproduce:

1. convert a `*.obj` file to `*.usd` using `source/standalone_examples/api/omni.kit.asset_converter/asset_usd_converter.py` with `use_meter_as_world_unit=True`
2. load in the object
3. play around with the orientation

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 24.04

### GPU Name

RTX 5080

### GPU Driver and CUDA versions

Driver 570.133.07, CUDA 12.8

michaellin6, 2025-07-11:
Hi @manavkulshrestha thank you for posting your issue. This seems like a bug, but I cannot replicate it on my side. Would you mind sharing your bottle asset? Below is what I got from trying some random obj and couldn't see the issue.

https://github.com/user-attachments/assets/ea15895e-efaf-48f2-bfb7-d31cee118628

manavkulshrestha, 2025-07-15:
Hi @michaellin6 , thank you for your response and current efforts. I've attached a zip file containing:

1. A minimal working example (`mwe.py`) which spawns in the bottle (just similarly to how I am spawning it in my use case). I ran it and can confirm I am encountering the same issue I described above for 5.0.0, but not for 4.5.0.
2. OBJ file for the bottle (see `meshes/` folder) and converted USD file for the bottle (see `usd/` folder)
3. Script I used to convert the obj file to usd (slight modifications from official example). This may not run since I just copy-pasted it from my own project's directory and it expects some other files and directories, but may be useful to look at

Please let me know if you need anything else from me to help with this. I'm currently using 4.5.0 but would like very much to upgrade to 5.0.0 since there's a rendering bug in the old version with regards to 50 series GPUs.

[issue_55.zip](https://github.com/user-attachments/files/21243575/issue_55.zip)

michaellin6, 2025-07-23:
Hi @manavkulshrestha, it seems like we have a bug on our side from the PhysxSchema, I will file a bug and hopefully we get that fixed soon. To unblock yourself for the moment, if you don't need to access the additional PhysX Rigid Body and Collider parameters, you can comment out the lines involving `PhysxSchema.PhysxCollisionAPI.Apply` and `PhysxSchema.PhysxRigidBodyAPI.Apply`. Thanks for sharing the minimal working example!

manavkulshrestha, 2025-07-24:
Hi @michaellin6 that's great to know, thanks. I just need collisions and to disable gravity for the object. I think this gives me physics collisions:
```
physx_collision_api = PhysxSchema.PhysxCollisionAPI.Apply(obj_prim)
physx_collision_api.CreateRestOffsetAttr(0.0)
```
and this removes gravity for the object?
```
physx_rigid_body_api = PhysxSchema.PhysxRigidBodyAPI.Apply(obj_prim)
physx_rigid_body_api.CreateDisableGravityAttr(not gravity)
```
Is there another way to achieve my desired properties? I'll be honest, I found the docs a bit confusing so it's possible some of the calls may be redundant. What is the difference between the `collision_api`, `physx_collision_api`, `mesh_collision_api`, `rigid_body_api`, and `physx_rigid_body_api`? Is there a good resource somewhere?

closed 2025-09-19 by michaellin6

ckennedy2050, 2025-09-29:
I am experiencing this issue as well. We've been able to work around it when we switch to triangle mesh colliders from convex hull. But this is not possible for many of our objects so I'd love to know if/when this bug is going to be addressed. Thanks!