isaac-sim/IsaacSim #58: [BUG]: ./build.sh fails with SSLError

fortmeier, 2025-07-02:
### Description

building the current main cbbda4f96aa0d1af96c1ac320cbf432fe105e928

fails with

>>> Fetching all dependencies.
Package 'kit-kernel' at version '107.3.0+feature.199947.b0a86421.gl.manylinux_2_35_x86_64.release' is missing from local storage.
packman(WARNING): SSLError: HTTPSConnectionPool(host='d2cerybs4newgl.cloudfront.net', port=443): Max retries exceeded with url: /kit-kernel%40107.3.0%2Bfeature.199947.b0a86421.gl.manylinux_2_35_x86_64.release.7z (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)')))
packman(WARNING): Retrying after 20 seconds (2 retries left) ...

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 22

### GPU Name

RTX 6000

### GPU Driver and CUDA versions

Driver Version: 570.144        CUDA Version: 12.8

juu7abt, 2025-07-23:
@sheikh-nv  i have the same problem on a similar workstation (same GPU but different CUDA versions)

even with the correct host d4i3qtqj3r0z5.cloudfront.net in the packman.config the SSL errors still show the wrong host:

`
>>> Fetching all dependencies.
Package 'kit-kernel' at version '107.3.0+feature.199947.b0a86421.gl.manylinux_2_35_x86_64.release' is missing from local storage.
packman(WARNING): SSLError: HTTPSConnectionPool(host='d2cerybs4newgl.cloudfront.net', port=443): Max retries exceeded with url: /kit-kernel%40107.3.0%2Bfeature.199947.b0a86421.gl.manylinux_2_35_x86_64.release.7z (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)')))
packman(WARNING): Retrying after 20 seconds (2 retries left) . 
`

So for anyone who needs a Quick solution:

- exchange the line 223 (`path_candidates.append(base_path + ".7z"`) and 224 (`path_candidates.append(base_path + ".zip"`) in file "~/.cache/packman/packman-common/7.28/packman/transport/httptransport.py" 
- rerun the build.sh and the kit-kernel .zip file will be downloaded from the correct host without changing the host anywhere, but the next dependency-download will fail with the same SSL error
- revert the first step of line-exchange and rerun the build.sh again, now it continues

Some findings:

Via some debug prints it can be confirmed that the download urls are all correct, even for kit-kernel, the problem seems to be the non-existence of the .7z file for the desired version:

`
>>> Fetching all dependencies.
packman(WARNING): Package at '/home/juu7abt/.cache/packman/chk/kit-kernel/107.3.0+feature.199947.b0a86421.gl.manylinux_2_35_x86_64.release' is corrupt (missing sha1 file) and no valid redirect was found
Package 'kit-kernel' at version '107.3.0+feature.199947.b0a86421.gl.manylinux_2_35_x86_64.release' is missing from local storage.
Remote name: packman:cloudfront with configs: {'packman:cloudfront': <packman.schemaparser.Remote object at 0x7fa8a99cbe10>}
Remote Transport: <packman.transport.httptransport.HttpTransport object at 0x7fa8a79dc190>
Checking remote 'packman:cloudfront' for package 'kit-kernel' at version '107.3.0+feature.199947.b0a86421.gl.manylinux_2_35_x86_64.release' ...
Searching for file at: 'http://d4i3qtqj3r0z5.cloudfront.net/kit-kernel%40107.3.0%2Bfeature.199947.b0a86421.gl.manylinux_2_35_x86_64.release.7z'
packman(WARNING): SSLError: HTTPSConnectionPool(host='d2cerybs4newgl.cloudfront.net', port=443): Max retries exceeded with url: /kit-kernel%40107.3.0%2Bfeature.199947.b0a86421.gl.manylinux_2_35_x86_64.release.7z (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)')))
packman(WARNING): Retrying after 20 seconds (2 retries left) ...
packman(WARNING): SSLError: HTTPSConnectionPool(host='d2cerybs4newgl.cloudfront.net', port=443): Max retries exceeded with url: /kit-kernel%40107.3.0%2Bfeature.199947.b0a86421.gl.manylinux_2_35_x86_64.release.7z (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)')))
packman(WARNING): Retrying after 20 seconds (1 retry left) ...

`

also ofc the SSLError msgs are wrong.... and causing confusions.

In comparison the .zip version of other dependencies are not available.... 

So IMHO the logic of checking the download candidates is not performing correctly, and it should drop an error after both .7z and .zip are checked instead of only the first one. Position of this logic: Function "    def get_download_info_for_package(
" in file "~/.cache/packman/packman-common/7.28/packman/transport/httptransport.py"

vick-yu, 2025-07-27:
We've created an internal ticket to track the SSL error issue you've been discussing. We appreciate you bringing this to our attention and providing your detailed findings.