isaac-sim/IsaacSim #59: [BUG]: Strange behaviour of annotation labels

TWeber132, 2025-07-03:
### Description

Hi, 

I tried to use the CocoWriter for synthetic data generation (extension: omni.replicator.core- ...), in an attempt to generate images from a rope made out of capsule prims.

But this happend:

[image]
[image]
[image]

The first image displays the data-field of the "instance_segmentation_fast" annotator, colorized.
The second  image displays the data-field of the "semantic_segmentation" annotator, also colorized.
The third image displays the data-field of the "instance_segmentation_fast" annotator, not colorized but masked according to the label from the annotator and a chosen color, background, other and unlabelled get masked black:
~~~
def _write_segmentation_image(self, render_product, camera_name, annotator_dict: dict):
        output_path = render_product + os.path.sep
        img_file_path = f"{output_path}panoptic_{camera_name}_{self.dataset_id}.{self._sequence_id}{self._frame_id:0{self._frame_padding}}.{self._image_output_format}"
        img_file_path_r = f"{output_path}panoptic_{camera_name}_{self.dataset_id}.{self._sequence_id}{self._frame_id:0{self._frame_padding}}_r.{self._image_output_format}"
        id_to_semantics = annotator_dict["idToSemantics"]
        # id_to_labels = annotator_dict["idToLabels"]

        # Create blank image
        colored_segmentation = np.zeros(
            (annotator_dict["data"].shape[0], annotator_dict["data"].shape[1], 4), dtype=np.uint8
        )
        # print(render_product, id_to_labels, id_to_semantics)
        for rgba_value, semantic_pair in id_to_semantics.items():
            # Looks similar to: '(0, 0, 0, 0)': {'class': 'BACKGROUND'}.
            for semantic_label in semantic_pair.values():
                try:
                    label_info = self.label_dict[semantic_label.lower()]
                except KeyError:
                    label_info = self.label_dict["other"]
                semantic_color = label_info["color"]

                if len(semantic_color) == 3:
                    semantic_color += (255,)  # Add alpha channel
                mask = np.all(annotator_dict["data"] == rgba_value, axis=2)
                colored_segmentation[mask] = semantic_color

        self.backend.schedule(F.write_image, data=colored_segmentation, path=img_file_path)
        self.backend.schedule(F.write_image, data=annotator_dict["data"], path=img_file_path_r)

        return img_file_path
~~~
For me it seems some weird things are happening (maybe only in my particular case) in a way that sometimes an entry is missing inside of  annotator_dict["idToSemantics"]. The same entry is also lost in id_to_labels = annotator_dict["idToLabels"], BUT appearently not in the colorized "instance_segmentation_fast" annotator?

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 24.04

### GPU Name

RTX 5090

### GPU Driver and CUDA versions

Driver 570.133.07 CUDA 12.8

zwdoescode, 2025-08-06:
Sorry for my late reply. Let me reach out to the proper person for this issue.

ahaidu-nv, 2025-08-07:
@TWeber132 does Link_13 have any other different properties compared to the other links?
What happens if you raise the camera a bit so that Link_13 is fully visible?
Have you noticed this issue in other scenarios as well?

I will try recreating this from your description scene.

JieFeng-cse, 2025-08-10:
@TWeber132 Hi Till, just out of curiosity, I want to ask, did you build the environment with gcc 11? I remember 5090 requires gcc 14 for drivers, but Isaac does not support it. Did you downgrade your gcc to 11 to make it work? Will that influence other stuff?
Can you give me some hints on how you build the environment for a 5090-based desktop? 
I know it is quite a few questions.
Really appreciate your potential help and suggestions for any of those questions.

TWeber132, 2025-08-10:
I think there is already a dockerfile inside a pull request, i guess you could also start from there.

Yes i used gcc 11 as specified by the README, it did not influence any of my other stuff, your mileage may vary.

This is my build process as part of the dockerfile i use:
```
FROM ubuntu:24.04 AS isaacsim_build
RUN userdel -r ubuntu

# Timezone
ENV TZ=Europe/Berlin
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone

# Share GPU
ENV NVIDIA_DRIVER_CAPABILITIES=all
ENV NVIDIA_VISIBLE_DEVICES=all

# Arguments for UID and GID
ARG UID=1000
ARG GID=1000
ARG USERNAME=omniverse_profi
RUN groupadd -g $GID ${USERNAME} && \
    useradd -m -l -u $UID -g ${USERNAME} ${USERNAME}

ENV USER=${USERNAME}

# Remove any third-party apt sources to avoid issues with expiring keys.
RUN rm -f /etc/apt/sources.list.d/*.list

# Avoid user inputs during installation process and choose defaults 
ARG DEBIAN_FRONTEND=noninteractive

# Download necessary packages for building isaacsim
RUN apt-get update -qq && apt-get install -y -qq git\
                                                 git-lfs \
                                                 build-essential \
                                                 gcc-11 \
                                                 g++-11 \
                                                 curl \
                                                 && rm -rf /var/lib/apt/lists/*
# Install dependencies for vulkan
RUN apt-get update -qq && apt-get install -y -qq pciutils \
                                                 vulkan-tools \ 
                                                 mesa-utils \
                                                 && rm -rf /var/lib/apt/lists/*
# Install dependencies for isaacsim
RUN apt-get update -qq && apt-get install -y -qq libglu1-mesa \
                                                 libxt6 \
                                                 && rm -rf /var/lib/apt/lists/*

# Set gcc and g++ to spcific version
RUN update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-11 200 && \
    update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-11 200

USER ${USER}
# Clone isaacsim repository
WORKDIR /home/${USER}
RUN git clone https://github.com/isaac-sim/IsaacSim.git isaacsim
# Choose specific commit, install git lfs and pull
WORKDIR /home/${USER}/isaacsim
RUN git checkout 494347a 
RUN git lfs install && git lfs pull
# Build isaacsim
RUN echo "yes" | ./build.sh

CMD [ "./isaac-sim.sh" ]
```
You'll also need NVIDIA Container Toolkit installed on your host machine, for you Docker to work.

If you don't want to use Docker installing as the README stated, worked in my case.

JieFeng-cse, 2025-08-10:
@TWeber132 Thak you so much for your detailed response!