isaac-sim/IsaacSim #61: [BUG]: OpenXR crash when connecting to Quest 3

wy-chien, 2025-07-04:
### Description

It crashes almost whenever trying to use Quest 3 with the OpenXR in the Isaac Sim.

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 22.04

### GPU Name

RTX 4090

### GPU Driver and CUDA versions

Driver 570.153 CUDA 12.8

### Logs

The connector I am using is `ALVR` plus `SteamVR` with runtime OpenXR enabled. 
This setup works fine with Isaac Sim 4.5.0, but does not with Isaac Sim 5.0.0. 

Whenever I click "StartVR", there is one frozen frame shown in the Quest3's ALVR app, and then it will crash after a few seconds.

Here's the log of the crash. Please let me know if this is too verbose, I can update the log file instead.

```bash
$ ./isaac-sim.xr.vr.sh 
...
[6.431s] app ready
2025-07-03T10:39:21Z [6,440ms] [Warning] [carb] Acquiring non optional plugin interface which is not listed as dependency: [carb::dictionary::ISerializer v1.1] (plugin: carb.dictionary.serializer-json.plugin), by client: carb.scenerenderer-rtx.plugin. Add it to CARB_PLUGIN_IMPL_DEPS() macro of a client.
2025-07-03T10:39:24Z [9,040ms] [Warning] [omni.kit.widget.filebrowser.thumbnails] Failed to stat url bookmarks:
2025-07-03T10:39:24Z [9,066ms] [Warning] [omni.kit.widget.filebrowser.thumbnails] Failed to stat url bookmarks:
2025-07-03T10:39:29Z [14,216ms] [Warning] [omni.graph.core.plugin] Could not find node type interface for 'isaacsim.ros2.bridge.ROS2SubscribeJointState'
2025-07-03T10:39:29Z [14,216ms] [Warning] [omni.graph.core.plugin] Could not find node type interface for 'isaacsim.ros2.bridge.ROS2Context'
2025-07-03T10:39:29Z [14,216ms] [Warning] [omni.graph.core.plugin] Could not find node type interface for 'isaacsim.ros2.bridge.ROS2SubscribeJointState'
2025-07-03T10:39:29Z [14,216ms] [Warning] [omni.graph.core.plugin] Could not find node type interface for 'isaacsim.ros2.bridge.ROS2SubscribeJointState'
2025-07-03T10:39:29Z [14,217ms] [Warning] [omni.graph.core.plugin] Could not find node type interface for 'isaacsim.ros2.bridge.ROS2PublishJointState'
2025-07-03T10:39:29Z [14,217ms] [Warning] [omni.graph.core.plugin] Could not find node type interface for 'isaacsim.ros2.bridge.ROS2SubscribeJointState'
2025-07-03T10:39:29Z [14,217ms] [Warning] [omni.graph.core.plugin] Could not find node type interface for 'isaacsim.ros2.bridge.ROS2SubscribeJointState'
2025-07-03T10:39:29Z [14,217ms] [Warning] [omni.graph.core.plugin] Could not find node type interface for 'isaacsim.ros2.bridge.ROS2SubscribeJointState'
2025-07-03T10:39:29Z [14,217ms] [Warning] [omni.graph.core.plugin] Could not find node type interface for 'isaacsim.ros2.bridge.ROS2PublishClock'
2025-07-03T10:39:29Z [14,235ms] [Warning] [omni.fabric.plugin] getAttributeCount called on non-existent path /sns_2f/sns_emily/Scene/wok_stand/Geometry/Wok_Stand_03_Top_1_2/CUBIC_CONNECTOR_45X45_3_3_55/_2023_06_20_13_54_09_0625_54
2025-07-03T10:39:29Z [14,235ms] [Warning] [omni.fabric.plugin] getTypes called on non-existent path /sns_2f/sns_emily/Scene/wok_stand/Geometry/Wok_Stand_03_Top_1_2/CUBIC_CONNECTOR_45X45_3_3_55/_2023_06_20_13_54_09_0625_54
2025-07-03T10:39:32Z [17,261ms] [Warning] [carb] Client gpu.foundation.plugin has acquired [gpu::unstable::IMemoryBudgetManagerFactory v0.1] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)
Error [GENERAL | xrCreateInstance | OpenXR-Loader] : LoaderInstance::CreateInstance chained CreateInstance call failed
Error [GENERAL | xrCreateInstance | OpenXR-Loader] : xrCreateInstance failed
2025-07-03T10:39:47Z [32,446ms] [Warning] [gpu.foundation.plugin] Invalid sync scope for texture resource 'RpResource XR Texture'. Create resource with valid sync scope for lifetime tracking or use kResourceUsageFlagNoSyncScope.
2025-07-03T10:39:47Z [32,446ms] [Warning] [gpu.foundation.plugin] Invalid sync scope for texture resource 'RpResource XR Texture'. Create resource with valid sync scope for lifetime tracking or use kResourceUsageFlagNoSyncScope.
2025-07-03T10:39:49Z [34,097ms] [Warning] [gpu.foundation.plugin] Invalid sync scope for texture resource 'RpResource XR Texture'. Create resource with valid sync scope for lifetime tracking or use kResourceUsageFlagNoSyncScope.
2025-07-03T10:39:49Z [34,097ms] [Warning] [gpu.foundation.plugin] Invalid sync scope for texture resource 'RpResource XR Texture'. Create resource with valid sync scope for lifetime tracking or use kResourceUsageFlagNoSyncScope.
2025-07-03T10:39:50Z [35,479ms] [Warning] [gpu.foundation.plugin] Invalid sync scope for texture resource 'RpResource XR Texture'. Create resource with valid sync scope for lifetime tracking or use kResourceUsageFlagNoSyncScope.
2025-07-03T10:39:50Z [35,479ms] [Warning] [gpu.foundation.plugin] Invalid sync scope for texture resource 'RpResource XR Texture'. Create resource with valid sync scope for lifetime tracking or use kResourceUsageFlagNoSyncScope.
2025-07-03T10:40:18Z [62,994ms] [Error] [omni.kit.xr.core.plugin] Viewport does not have the right aov
2025-07-03T10:40:18Z [62,995ms] [Warning] [omni.kit.xr.core.plugin] [XR Composition pipeline] Unwarp/resample failed to execute: XRComposition (warped) input resource is null
2025-07-03T10:40:18Z [62,995ms] [Warning] [omni.kit.xr.core.plugin] [XR Composition pipeline] Unwarp/resample failed to execute: XRComposition (warped) input resource is null
2025-07-03T10:40:18Z [63,021ms] [Warning] [omni.kit.xr.core.plugin] [XR Composition pipeline] Unwarp/resample failed to execute: XRComposition (warped) input resource is null
2025-07-03T10:40:18Z [63,021ms] [Warning] [omni.kit.xr.core.plugin] [XR Composition pipeline] Unwarp/resample failed to execute: XRComposition (warped) input resource is null
2025-07-03T10:40:18Z [63,049ms] [Warning] [omni.kit.xr.core.plugin] [XR Composition pipeline] Unwarp/resample failed to execute: XRComposition (warped) input resource is null
2025-07-03T10:40:18Z [63,049ms] [Warning] [omni.kit.xr.core.plugin] [XR Composition pipeline] Unwarp/resample failed to execute: XRComposition (warped) input resource is null
2025-07-03T10:40:18Z [63,075ms] [Warning] [omni.kit.xr.core.plugin] [XR Composition pipeline] Unwarp/resample failed to execute: XRComposition (warped) input resource is null
2025-07-03T10:40:18Z [63,075ms] [Warning] [omni.kit.xr.core.plugin] [XR Composition pipeline] Unwarp/resample failed to execute: XRComposition (warped) input resource is null
2025-07-03T10:40:18Z [63,105ms] [Warning] [omni.kit.xr.core.plugin] [XR Composition pipeline] Unwarp/resample failed to execute: XRComposition (warped) input resource is null
2025-07-03T10:40:18Z [63,105ms] [Warning] [omni.kit.xr.core.plugin] [XR Composition pipeline] Unwarp/resample failed to execute: XRComposition (warped) input resource is null
2025-07-03T10:40:18Z [63,130ms] [Warning] [omni.kit.xr.core.plugin] [XR Composition pipeline] Unwarp/resample failed to execute: XRComposition (warped) input resource is null
2025-07-03T10:40:18Z [63,130ms] [Warning] [omni.kit.xr.core.plugin] [XR Composition pipeline] Unwarp/resample failed to execute: XRComposition (warped) input resource is null
2025-07-03T10:40:18Z [63,149ms] [Warning] [omni.kit.xr.core.plugin] [XR Composition pipeline] Unwarp/resample failed to execute: XRComposition (warped) input resource is null
[... 538 lines cut ...]
'
2025-07-03T10:40:19Z [80ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]   'telemetrySessionId' = '12761392439191396805'
2025-07-03T10:40:19Z [81ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]   'terminatedByAbort' = '0'
2025-07-03T10:40:19Z [81ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]   'totalRamBareMetalMB' = '63990'
2025-07-03T10:40:19Z [81ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]   'totalRamLimitedMB' = '63990'
2025-07-03T10:40:19Z [82ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]   'totalSwapBareMetalMB' = '2047'
2025-07-03T10:40:19Z [82ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]   'totalSwapLimitedMB' = '2047'
2025-07-03T10:40:19Z [83ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]   'userId' = 'default'
2025-07-03T10:40:19Z [83ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]   'workingDirectory' = '/home/<username>/IsaacSim/_build/linux-x86_64/release'
2025-07-03T10:40:19Z [83ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash] Crash report files for upload:
[... cut ...]
2025-07-03T10:40:21Z [1,983ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash] Deleting file: '/home/shinfang-ovx/.local/share/ov/data/Kit/Isaac-Sim XR VR/5.0/db949c17-e46d-4df1-dad28386-81f1adf3.py.txt'
2025-07-03T10:40:21Z [1,984ms] [Info] [carb.crashreporter-breakpad.plugin] attempting to delete the file '/home/shinfang-ovx/.local/share/ov/data/Kit/Isaac-Sim XR VR/5.0/db949c17-e46d-4df1-dad28386-81f1adf3.py.txt' with 10 retries remaining.
2025-07-03T10:40:21Z [1,984ms] [Error] [carb.crashreporter-breakpad.plugin] [crash]     dump file size is 0 bytes, file is readable.
```

zwdoescode, 2025-07-22:
@wy-chien I don't have good knowledge/information on this. Let me follow up with the internal team.

wy-chien, 2025-07-29:
@zwdoescode Sounds good, looking forward to any possible solutions or workarounds.

zwdoescode, 2026-04-17:
Sorry for my late reply. The fix is merged and published with Kit XR exts 107.3.108

closed 2026-04-17 by zwdoescode