isaac-sim/IsaacSim #62: [BUG]: The download link in the documentation for version 5.0.0 still points to version 4.5.0

zhxchen-se, 2025-07-07:
### Description

[image]
If the download is currently unavailable, does that mean the only way to install 5.0.0 is by building it from the GitHub source code? Thanks for looking into it.

### Isaac Sim version

-

### Operating System (OS)

-

### GPU Name

-

### GPU Driver and CUDA versions

-

zwdoescode, 2025-07-07:
Thanks for pointing this out! I will ask the internal team to fix this

liilatorabi, 2025-07-08:
That is correct. We are in the development phase, the final build will be available end of July when Isaac Sim 5.0 is released.