isaac-sim/IsaacSim #63: Failed to import isaacsim.asset.gen.omap

dongzhue, 2025-07-07:
### Description

I want to use a stand alone python script to automatically generate an Occupancy Map.
I refer to the example provided by the official documentation, but I **cannot import the isaacsim.asset library**. I found that this _omap is located in the ext folder of isaacsim. 
I made sure I was using the correct conda environment and was able to calculate the occ map in the GUI。
How can I correctly run the example provided by the official website?

### Isaac Sim version

4.5.0

### Operating System (OS)

Ubuntu 22.04

### GPU Name

RTX 4090 D

### GPU Driver and CUDA versions

CUDA 12.4

### Logs

**Reference code：**
[image]
**Error message：**
[image]

hmazhar-nv, 2025-07-07:
Thanks for reporting, it should be:
```from isaacsim.asset.gen.omap.bindings import _omap```

Will fix the docs, in the meantime, you can see how we use it in our tests here:
https://github.com/isaac-sim/IsaacSim/blob/main/source/extensions/isaacsim.asset.gen.omap/python/tests/test_occupancy.py

dongzhue, 2025-07-08:
Thank you very much！

dongzhue, 2025-07-08:
Thanks for your advice, I found that I can now successfully run the example code on the script editor, but when I run it as a stand alone script, I still encounter the problem of '**ModuleNotFoundError: No module named 'isaacsim.asset**''. Is this because my environment is not imported correctly?My Isaac sim was installed using pip.
 Do you have any suggestions? Thank you.

[image]

dongzhue, 2025-07-08:
> Add the following after SimulationApp() starts. By default to reduce startup time we don't have all extensions enabled for standalone workflows在 SimulationApp（） 启动后添加以下内容。默认情况下，为了减少启动时间，我们没有为独立工作流启用所有扩展
> 
> ```
> from isaacsim.core.utils.extensions import enable_extension
> enable_extension("isaacsim.asset.gen.omap")
> ```

Got it, thanks!