isaac-sim/IsaacSim #69: Default install has issue with pip: numpy.dtype size changed

jgillick, 2025-07-10:
### Description

Stock cloud installation, following the steps, generates this error (among others) when trying to run the example RL command.

```
2025-07-10 21:42:18 [11,579ms] [Error] [omni.ext._impl._internal] Failed to import python module isaacsim.core.simulation_manager from /isaac-sim/exts/isaacsim.core.simulation_manager. Error: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject. Traceback:
Traceback (most recent call last):
```

It appears that the wrong version of numpy is being installed.

```
A NumPy version >=1.19.5 and <1.27.0 is required for this version of SciPy (detected version 2.2.6)
```

### Steps to reproduce

* I following the [cloud installation steps](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/cloud_installation.html), I installed Isaac Sim to AWS using all the defaults (us-east-1, g5.2xlarge, etc). 
* I logged into the machine using NoMachine
* Double-clicked the "isaaclab.sh" icon on the desktop.
* Inside the Iassac Lab terminal, I ran the example command from the [cloud installation docs](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/cloud_installation.html): `./isaaclab.sh -p scripts/reinforcement_learning/rl_games/train.py --task=Isaac-Cartpole-v0`
* Isaac Sim opens (after a couple "This application is not responding" warning), but it has nothing loaded and nothing responds when clicking.
* Going back to the terminal, I see the a lot of errors about numpy

### Isaac Sim version

4.5.0

### Operating System (OS)

Ubuntu 22.04.5 LTS

### GPU Name

GA102GL / AWS g5.2xlarge

### GPU Driver and CUDA versions

Driver Version: 535.247.01   CUDA Version: 12.2

### Logs

```txt
Welcome to Isaac Lab!

[IsaacLab]:/isaaclab$ ./isaaclab.sh -p scripts/reinforcement_learning/rl_games/train.py --task=Isaac-Cartpole-v0 > /error_log.txt
/isaac-sim/exts/omni.pip.compute/pip_prebundle/scipy/__init__.py:143: UserWarning: A NumPy version >=1.19.5 and <1.27.0 is required for this version of SciPy (detected version 2.2.6)
  warnings.warn(f"A NumPy version >={np_minversion} and <{np_maxversion}"
2025-07-10 22:01:53 [11,005ms] [Error] [omni.ext._impl._internal] Failed to import python module isaacsim.core.simulation_manager from /isaac-sim/exts/isaacsim.core.simulation_manager. Error: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject. Traceback:
Traceback (most recent call last):
  File "/isaaclab/_isaac_sim/kit/kernel/py/omni/ext/_impl/_internal.py", line 187, in _custom_importer
    return _import_public(ext_module.path, ext_module.name, reload_enabled)
  File "/isaaclab/_isaac_sim/kit/kernel/py/omni/ext/_impl/_internal.py", line 94, in _import_public
    module = import_module(module_name)
  File "/isaaclab/_isaac_sim/kit/kernel/py/omni/ext/_impl/custom_importer.py", line 76, in import_module
    return importlib.import_module(name)
  File "/isaaclab/_isaac_sim/kit/python/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/isaac-sim/exts/isaacsim.core.simulation_manager/isaacsim/core/simulation_manager/__init__.py", line 1, in <module>
    from .impl.extension import Extension
  File "/isaac-sim/exts/isaacsim.core.simulation_manager/isaacsim/core/simulation_manager/impl/__init__.py", line 1, in <module>
    from .extension import *
  File "/isaac-sim/exts/isaacsim.core.simulation_manager/isaacsim/core/simulation_manager/impl/extension.py", line 7, in <module>
    from .simulation_manager import SimulationManager
[... 746 lines cut ...]
    import isaacsim.core.utils.numpy as np_utils
  File "/isaac-sim/exts/isaacsim.core.utils/isaacsim/core/utils/numpy/__init__.py", line 10, in <module>
    from .rotations import *
  File "/isaac-sim/exts/isaacsim.core.utils/isaacsim/core/utils/numpy/rotations.py", line 13, in <module>
    from scipy.spatial.transform import Rotation
  File "/isaac-sim/exts/omni.pip.compute/pip_prebundle/scipy/spatial/__init__.py", line 105, in <module>
    from ._kdtree import *
  File "/isaac-sim/exts/omni.pip.compute/pip_prebundle/scipy/spatial/_kdtree.py", line 4, in <module>
    from ._ckdtree import cKDTree, cKDTreeNode
  File "_ckdtree.pyx", line 1, in init scipy.spatial._ckdtree
ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject

2025-07-10 22:02:05 [22,869ms] [Error] [carb.scripting-python.plugin] Exception: Extension python module: 'isaaclab_tasks' in '/isaaclab/source/isaaclab_tasks' failed to load.

At:
  /isaaclab/_isaac_sim/kit/kernel/py/omni/ext/_impl/_internal.py(213): startup
  /isaaclab/_isaac_sim/kit/kernel/py/omni/ext/_impl/_internal.py(328): startup_extension
  PythonExtension.cpp::startup()(2): <module>
  /isaaclab/_isaac_sim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py(427): _start_app
  /isaaclab/_isaac_sim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py(212): __init__
  /isaaclab/source/isaaclab/isaaclab/app/app_launcher.py(780): _create_app
  /isaaclab/source/isaaclab/isaaclab/app/app_launcher.py(131): __init__
  /isaaclab/scripts/reinforcement_learning/rl_games/train.py(42): <module>

```

jgillick, 2025-07-10:
Additionally: after executing the isaaclab.sh program on the desktop, I tried to reinstall things with `./isaaclab.sh -i` and that did not fix the issue.

zwdoescode, 2025-07-14:
@jgillick Have you tried downgrading the NumPy version?

jgillick, 2025-07-15:
When installing a newer version of NumPy, it seems to work. It seems like this should be handled by the installer, though.

zwdoescode, 2025-07-15:
Thanks for your feedback @jgillick! As for the installer, are you talking about IsaacAutomator

jgillick, 2025-07-19:
That's a good question, I'm not really sure where this is introduced. It could very well be on the IsaacAutomator side, since that was what setup the server. I assumed that the IsaacAutomator just calls the standard installation scripts in Isaac sim/lab.

dxyy1, 2025-07-20:
I ran into the same issue, and the command below resolved the error for me, though there does seem to be some error/warning when downgraded it.
'''
root@I2R-SPD-7150124:/workspace/isaaclab/scripts/demos# pip install --force-reinstall numpy==1.26.4
Collecting numpy==1.26.4
  Downloading numpy-1.26.4-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (61 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.0/61.0 kB 2.0 MB/s eta 0:00:00
Downloading numpy-1.26.4-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (18.2 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 18.2/18.2 MB 164.4 MB/s eta 0:00:00
Installing collected packages: numpy
  Attempting uninstall: numpy
    Found existing installation: numpy 2.2.6
    Uninstalling numpy-2.2.6:
      Successfully uninstalled numpy-2.2.6
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
pin-pink 3.1.0 requires quadprog>=0.1.11, which is not installed.
nvidia-srl-usd 0.19.1 requires usd-core<24.00,>=21.11, which is not installed.
nvidia-srl-usd-to-urdf 0.7.2 requires usd-core<24.00,>=21.11, which is not installed.
opencv-python 4.12.0.88 requires numpy<2.3.0,>=2; python_version >= "3.9", but you have numpy 1.26.4 which is incompatible.
nvidia-srl-usd-to-urdf 0.7.2 requires lxml<5.0.0,>=4.9.2, but you have lxml 5.3.0 which is incompatible.
Successfully installed numpy-1.26.4
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

[notice] A new release of pip is available: 24.0+nv1 -> 25.1.1
[notice] To update, run: /workspace/isaaclab/_isaac_sim/kit/python/bin/python3 -m pip install --upgrade pip
'''

luoye2333, 2025-07-28:
`pip install --force-reinstall numpy==1.26.4` works, thank you @dxyy1 . The problem seems to be in the prebuilt numpy.

My log:
```
  File "/home/u001/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/numpy/random/_pickle.py", line 1, in <module>
    from .mtrand import RandomState
  File "numpy/random/mtrand.pyx", line 1, in init numpy.random.mtrand
ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject
```