isaac-sim/IsaacSim #80: [BUG]: Build failed on Ubuntu 24.04 with no intuitive error message

kevin1zc, 2025-07-19:
### Description

Followed instructions in README on Ubuntu 24.04 and used gcc 11.5.0, but build failed with exit code 1 and empty error message. Log file is attached.

One thing might be helpful: I am using Lmod to manage different versions of gcc and other packages.

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 24.04

### GPU Name

RTX 4090

### GPU Driver and CUDA versions

Driver Version: 570.169        CUDA Version: 12.9

### Logs

[repo.log](https://github.com/user-attachments/files/21325163/repo.log)

kevin1zc, 2025-07-21:
This message comes from the python executable. I'm assuming it only indicates where python was compiled? I'm using Ubuntu 24.04, not Red Hat, and I don't have GCC 7.3.1 either.

[image]

hmazhar-nv, 2025-07-22:
Can you try running the command that failed, does it provide any additional output?

`/home/zhuochen/Softwares/isaacsim/_build/linux-x86_64/release/kit/kit /home/zhuochen/Softwares/isaacsim/_build/linux-x86_64/release/apps/isaacsim.exp.extscache.kit --allow-root --portable --ext-precache-mode --/crashreporter/gatherUserStory=0 --/app/settings/persistent=0 --/app/settings/loadUserConfig=0 --/app/extensions/generateVersionLock=1 --/app/extensions/parallelPullEnabled=1 --/app/enableStdoutOutput=1 --/app/extensions/detailedSolverExplanation=1 --/app/extensions/registryEnabled=1 --/app/extensions/mkdirExtFolders=0 '--/app/extensions/registryCacheFull='"'"'/home/zhuochen/Softwares/isaacsim/_build/linux-x86_64/release/extscache'"'"'' --/log/flushStandardStreamOutput=1 '--/exts/omni.kit.registry.nucleus/registries/0/name="kit/default"' '--/exts/omni.kit.registry.nucleus/registries/0/url="omniverse://kit-extensions.ov.nvidia.com/exts/kit/default"' '--/exts/omni.kit.registry.nucleus/registries/1/name="kit/sdk"' '--/exts/omni.kit.registry.nucleus/registries/1/url="omniverse://kit-extensions.ov.nvidia.com/exts/kit/sdk/${kit_version_short}/${kit_git_hash}"' --/app/extensions/target/config=release --ext-folder /home/zhuochen/Softwares/isaacsim/_build/linux-x86_64/release/exts --ext-folder /home/zhuochen/Softwares/isaacsim/_build/linux-x86_64/release/extsDeprecated`

closed 2025-07-22 by kevin1zc