isaac-sim/IsaacSim #81: undefined symbol: __nvJitLinkCreate_12_8, version libnvJitLink.so.12

belalhmedan90, 2025-07-21:
### Description

```bash
2025-07-21T19:08:11Z [7,040ms] [Error] [omni.ext._impl.custom_importer] Failed to import python module isaacsim.core.simulation_manager. Error: /home/belal.hmedan/projects/isaacsim/_build/linux-x86_64/release/exts/omni.isaac.ml_archive/pip_prebundle/torch/lib/../../nvidia/cusparse/lib/libcusparse.so.12: undefined symbol: __nvJitLinkCreate_12_8, version libnvJitLink.so.12. Traceback:
Traceback (most recent call last):
  File "/home/belal.hmedan/.cache/packman/chk/kit-kernel/107.3.0+feature.199947.b0a86421.gl.manylinux_2_35_x86_64.release/kernel/py/omni/ext/_impl/custom_importer.py", line 85, in import_module
    return importlib.import_module(name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/belal.hmedan/.cache/packman/chk/kit-kernel/107.3.0+feature.199947.b0a86421.gl.manylinux_2_35_x86_64.release/python/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/belal.hmedan/projects/isaacsim/_build/linux-x86_64/release/exts/isaacsim.core.simulation_manager/isaacsim/core/simulation_manager/__init__.py", line 15, in <module>
    from .impl.extension import Extension
  File "/home/belal.hmedan/projects/isaacsim/_build/linux-x86_64/release/exts/isaacsim.core.simulation_manager/isaacsim/core/simulation_manager/impl/__init__.py", line 15, in <module>
    from .extension import *
  File "/home/belal.hmedan/projects/isaacsim/_build/linux-x86_64/release/exts/isaacsim.core.simulation_manager/isaacsim/core/simulation_manager/impl/extension.py", line 21, in <module>
    from .simulation_manager import SimulationManager
  File "/home/belal.hmedan/projects/isaacsim/_build/linux-x86_64/release/exts/isaacsim.core.simulation_manager/isaacsim/core/simulation_manager/impl/simulation_manager.py", line 20, in <module>
    import isaacsim.core.utils.torch as torch_utils
  File "/home/belal.hmedan/projects/isaacsim/_build/linux-x86_64/release/exts/isaacsim.core.utils/isaacsim/core/utils/torch/__init__.py", line 15, in <module>
    from .maths import *
  File "/home/belal.hmedan/projects/isaacsim/_build/linux-x86_64/release/exts/isaacsim.core.utils/isaacsim/core/utils/torch/maths.py", line 19, in <module>
    import torch
  File "/home/belal.hmedan/projects/isaacsim/_build/linux-x86_64/release/exts/omni.isaac.ml_archive/pip_prebundle/torch/__init__.py", line 409, in <module>
    from torch._C import *  # noqa: F403
    ^^^^^^^^^^^^^^^^^^^^^^
ImportError: /home/belal.hmedan/projects/isaacsim/_build/linux-x86_64/release/exts/omni.isaac.ml_archive/pip_prebundle/torch/lib/../../nvidia/cusparse/lib/libcusparse.so.12: undefined symbol: __nvJitLinkCreate_12_8, version libnvJitLink.so.12

2025-07-21T19:08:11Z [7,041ms] [Error] [carb.scripting-python.plugin] Exception: Extension python module: 'isaacsim.core.simulation_manager' in '/home/belal.hmedan/projects/isaacsim/_build/linux-x86_64/release/exts/isaacsim.core.simulation_manager' failed to load.

At:
  /home/belal.hmedan/.cache/packman/chk/kit-kernel/107.3.0+feature.199947.b0a86421.gl.manylinux_2_35_x86_64.release/kernel/py/omni/ext/_impl/_internal.py(222): startup
  /home/belal.hmedan/.cache/packman/chk/kit-kernel/107.3.0+feature.199947.b0a86421.gl.manylinux_2_35_x86_64.release/kernel/py/omni/ext/_impl/_internal.py(337): startup_extension
  PythonExtension.cpp::startup()(2): <module>

2025-07-21T19:08:11Z [7,041ms] [Error] [omni.ext.plugin] [ext: isaacsim.core.simulation_manager-1.2.2] Failed to startup python extension.
```

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 22.04

### GPU Name

NVIDIA RTX 4000 SFF Ada

### GPU Driver and CUDA versions

575.64.03

### Additional information

I cloned latest commmit and built according to instructions !

closed 2025-07-21 by belalhmedan90

csjiyw, 2025-07-25:
i use sudo apt install cuda-toolkit-12-8 to resolve this issue

belalhmedan90, 2025-07-25:
> i use `sudo apt install cuda-toolkit-12-8` to resolve this issue

I have installed whole system based on 12-6 ! I can unset the path as mentioned up and it works