isaac-sim/IsaacSim #88: Scan ghost obstacles in ROS 2 Navigation example

Cdfghglz, 2025-07-26:
### Description

Running the [ROS 2 Navigation example](https://docs.isaacsim.omniverse.nvidia.com/latest/ros2_tutorials/tutorial_ros2_navigation.html) I am unable to reproduce the navigation due to false positives obstacles in the local costmap:

[image]

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 22.04

### GPU Name

RTX 4060 mobile

### GPU Driver and CUDA versions

Driver Version: 570.133.07     CUDA Version: 12.8

zwdoescode, 2025-07-28:
@Cdfghglz are you using the nova carter example from the tutorial or are you using your customized scene and robot?

Cdfghglz, 2025-07-29:
The example loaded from the Sim GUI under Robotics Examples + IsaacSim-ros_workspaces @ 602752b.

I tried on multiple machines, laptop i7 + 4060 mobile as well as AWS g6e with 8vCPUs. On AWS it seemed a little better but still:

[image]