isaac-sim/IsaacSim #90: undefined symbol

driven-by-code, 2025-07-28:
### Description

Hello Everyone , 
I have been getting the same bug mentioned in this existing issues:
https://github.com/isaac-sim/IsaacSim/issues/24#issuecomment-2987405347
https://github.com/isaac-sim/IsaacSim/issues/20.
I tried upgrading to driver version 575 as well before downgrading back to 570. 
I have done `unset LD_LIBRARY_PATH`.
I have cuda 12.8 installed and also tried

```bash
export LD_LIBRARY_PATH=/usr/local/cuda-12.8/lib64:{$LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}
export PATH=/usr/local/cuda-12.8/bin:{$PATH:+:${PATH}}
export CUDA_HOME=/usr/local/cuda-12.8
```.
I'm still getting this issue and I have tried rebuilding fresh as well.
Pleas let me know if I am missing anything. 

### Isaac Sim version

5.0.0

### Operating System (OS)

22.04 

### GPU Name

RTX 5000 Ada Laptop GPU 

### GPU Driver and CUDA versions

570.158.01

driven-by-code, 2025-07-28:
I can provide the full output logs.
Do you want me to provide the torch cuda version after running `./python.sh` or the one installed on the machine ?If it is the machine, I do not have it installed. 
Thank you .

driven-by-code, 2025-07-29:
Here is the output of the torch cuda version:\

```bash
>>> print(torch.version.cuda)
12.4
```
Running `ldd` on the library , could you please elaborate how could I do it for the library ? 
Thank you !

vick-yu, 2025-07-29:
> Here is the output of the torch cuda version:
> 
> \>\>\> print(torch.version.cuda)
> 12.4

The version is incorrect. 

> Running `ldd` on the library , could you please elaborate how could I do it for the library ?

`ldd /home/user/isaacsim/_build/linux-x86_64/release/exts/omni.isaac.ml_archive/pip_prebundle/torch/lib/../../nvidia/cusparse/lib/libcusparse.so.12`

driven-by-code, 2025-07-29:
> The version is incorrect.
>
>
> `ldd /home/user/isaacsim/_build/linux-x86_64/release/exts/omni.isaac.ml_archive/pip_prebundle/torch/lib/../../nvidia/cusparse/lib/libcusparse.so.12`

Thank you !
 If the version of cuda that torch is using is incorrect , do you have an idea of what could be causing it to use the wrong cuda version ? 

```bash
ldd /home/user/isaacsim/_build/linux-x86_64/release/exts/omni.isaac.ml_archive/pip_prebundle/torch/lib/../../nvidia/cusparse/lib/libcusparse.so.12
	linux-vdso.so.1 (0x00007ffd97b82000)
	libnvJitLink.so.12 => /home/user/isaacsim/_build/linux-x86_64/release/exts/omni.isaac.ml_archive/pip_prebundle/torch/lib/../../nvidia/cusparse/lib/../../nvjitlink/lib/libnvJitLink.so.12 (0x000073bc4c200000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x000073bc698f0000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x000073bc698eb000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x000073bc698e6000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x000073bc51d19000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x000073bc698c4000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x000073bc4be00000)
	/lib64/ld-linux-x86-64.so.2 (0x000073bc6991f000)

```

vick-yu, 2025-07-29:
Could you please clarify what command is being run when you encounter this issue?

driven-by-code, 2025-07-29:
> `/home/user/.nvidia-omniverse/logs/Kit/Isaac-Sim Full/5.0/kit_20250729_080431.log`

The log file does not exist at that locatio.

driven-by-code, 2025-07-29:
My apologies.Here it is.
[kit_20250729_080431.log](https://github.com/user-attachments/files/21493911/kit_20250729_080431.log)

driven-by-code, 2025-07-29:
> `/home/user/isaacsim/_build/linux-x86_64/release/python.sh` environment?

Yes it is.
I tried rebuilding and running using the above mentioned command . I still have the same error sadly. 
Thank you for your responses !

hmazhar-nv, 2025-07-29:
can you run `./python.sh -m pip show torch`

it should output something like:
```
./python.sh -m pip show torch
Name: torch
Version: 2.7.0+cu128
Summary: Tensors and Dynamic neural networks in Python with strong GPU acceleration
Home-page: https://pytorch.org/
Author: PyTorch Team
Author-email: packages@pytorch.org
License: BSD-3-Clause
Location: /home/user/repos/omni_isaac_sim/_build/linux-x86_64/release/exts/omni.isaac.ml_archive/pip_prebundle
Requires: filelock, fsspec, jinja2, networkx, nvidia-cublas-cu12, nvidia-cuda-cupti-cu12, nvidia-cuda-nvrtc-cu12, nvidia-cuda-runtime-cu12, nvidia-cudnn-cu12, nvidia-cufft-cu12, nvidia-cufile-cu12, nvidia-curand-cu12, nvidia-cusolver-cu12, nvidia-cusparse-cu12, nvidia-cusparselt-cu12, nvidia-nccl-cu12, nvidia-nvjitlink-cu12, nvidia-nvtx-cu12, sympy, triton, typing-extensions
Required-by: torchaudio, torchvision
```

driven-by-code, 2025-07-29:
> ./python.sh -m pip show torch
```bash
Name: torch
Version: 2.5.1
Summary: Tensors and Dynamic neural networks in Python with strong GPU acceleration
Home-page: https://pytorch.org/
Author: PyTorch Team
Author-email: packages@pytorch.org
License: BSD-3-Clause
Location: /home/user/isaacsim/_build/linux-x86_64/release/kit/python/lib/python3.11/site-packages
Requires: filelock, fsspec, jinja2, networkx, nvidia-cublas-cu12, nvidia-cuda-cupti-cu12, nvidia-cuda-nvrtc-cu12, nvidia-cuda-runtime-cu12, nvidia-cudnn-cu12, nvidia-cufft-cu12, nvidia-curand-cu12, nvidia-cusolver-cu12, nvidia-cusparse-cu12, nvidia-nccl-cu12, nvidia-nvjitlink-cu12, nvidia-nvtx-cu12, sympy, triton, typing-extensions
Required-by: torchaudio, torchvision
```

driven-by-code, 2025-07-29:
I can . I ran this on an fresh build compiled from a fresh clone as well.

hmazhar-nv, 2025-07-29:
Actually better solution:
rm -m ~/.cache/packman/chk/kit-kernel/
and then build.sh again

if you want to be conservative delete the specific one that is symlinked to `/home/user/isaacsim/_build/linux-x86_64/release/kit`

This will force the kit-kernel packman dep to redownload, effectively deleting the `/home/user/isaacsim/_build/linux-x86_64/release/kit` folder and the site-packages inside of it

driven-by-code, 2025-07-29:
> Actually, better solution: rm -m ~/.cache/packman/chk/kit-kernel/ and then build.sh again
>
> if you want to be conservative delete the specific one that is symlinked to [...]

I tried this , it did not change anything. 
I deleted the whole `packman` folder and did a build from a fresh clone again.
Here is the log @vick-yu 
Thank you once again ! 
[build_log.txt](https://github.com/user-attachments/files/21496338/build_log.txt)