isaac-sim/IsaacSim #103: Isaac Sim WebRTC Streaming Client NOT work on Mac Apple M2

gujiacheng, 2025-08-04:
### Description

isaac sim version 4.5.0, installed by docker container
client version [macOS x86_64](https://download.isaacsim.omniverse.nvidia.com/isaacsim-webrtc-streaming-client-1.0.6-macos-x64.dmg) (208.2 MB)

[image]

but the client shows nothing after click the connect button

### Isaac Sim version

4.5.0

### Operating System (OS)

22.04

### GPU Name

A800

### GPU Driver and CUDA versions

isaac sim 4.5.0 official docker image

gujiacheng, 2025-08-05:
> Thank you for reporting this issue. It seems you are using the client for macOS x86_64 architecture, which is intended for Macs with Intel processors. Since your Mac has an Apple M2 chip, it uses the ARM64 architecture.
> 
> Additionally, Isaac Sim 5.0 is now open source. I recommend using the latest version of Isaac Sim to get the newest security patches and bug fixes.

switch to ARM64 architecture , still the same: 

[image]

reply later after trying latest version of Isaac sim

gujiacheng, 2025-08-05:
can not build due to error:
'' Package 'repo_man' at version '1.86.0' is missing from local storage. ''

[image]

sheikh-nv, 2025-08-12:
Hi @gujiacheng, on the [datasheet](https://resources.nvidia.com/en-us-briefcase-for-datasheets/proviz-a800-40gb-dat?ncid=no-ncid), the A800 do not have the NVENC support listed. Livestreaming may not work with this GPU when hosting Isaac Sim app.