isaac-sim/IsaacSim #108: IsaacSim 5.0 Gaussian Splatting Rendering Has Wrong Occlusion

nancywan1004, 2025-08-05:
### Description

Hi there,

I encountered an issue when I tried to import and render two 3DGRUT converted USDZ files containing 3D Gaussian Splatting data. The occulusion is wrong from the depth perspective, where the further one is cutting off the nearer one as you can tell from the pictures attached. 

On the other hand, I had another discovery where the performance would drop from a further distance above the scene, but would be fine if I zoom in, with the whole scene rendered in the camera frustum at the whole time. Any ideas on what could be the reason here? 

Thanks in advance!

[image]
[image]

 

### Isaac Sim version

5.0.0

### Operating System (OS)

Windows 10.19045

### GPU Name

RTX 3080

### GPU Driver and CUDA versions

Driver 566.24, CUDA 12.7

zwdoescode, 2025-08-12:
Thanks @nancywan1004! Here is how it looks like on my side. There are some gaps between the four imported USDZ. Other than that, it seems normal... Do you see any abnormal behavior?

[image]

zwdoescode, 2025-08-12:
Here is a screen recording that I move the rendered biker around
https://github.com/user-attachments/assets/2572c8d8-47f8-48a3-b3c2-42a2be3e95be

zwdoescode, 2026-01-14:
The internal team has a fix for this issue. The fix will be included in Isaac Sim 6.1. Thanks for your patience and please stay tuned!