isaac-sim/IsaacSim #110: Cannot retrieve RTX lidar data using headless Python script

simplelife2010, 2025-08-05:
### Description

I am attaching a USD stage and a Python script for reproduction:
[Archive.zip](https://github.com/user-attachments/files/21603179/Archive.zip)

In this simplified setup I have a rotary RTX lidar and an object in the stage. I want to use a python script to retrieve the lidar data for every simulation step. When using headless=True in the simulation app configuration, I get only empty lists for each step:

```
...
[11.960s] Simulation App Startup Complete
2025-08-05T15:48:03Z [13,079ms] [Warning] [usdrt.hydra.fabric_scene_delegate.plugin] using high frequency span with attrs is disabled
Step 0
2025-08-05T15:48:03Z [13,642ms] [Warning] [omni.timeline.plugin] Deprecated: direct use of ITimeline callbacks is deprecated. Use ITimeline::getTimeline (Python: omni.timeline.get_timeline_interface) instead.
Lidar: []
Step 1
2025-08-05T15:48:03Z [13,789ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
Lidar: []
Step 2
Lidar: []
Step 3
Lidar: []
Step 4
Lidar: []
Step 5
Lidar: []
Step 6
Lidar: []
Step 7
Lidar: []
Step 8
Lidar: []
Step 9
Lidar: []
2025-08-05T15:48:05Z [15,398ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
[15.522s] Simulation App Shutting Down
```

Interestingly, I get data beginning from step 6, if I set headless to False:
```
Step 3
Lidar: []
Step 4
Lidar: []
Step 5
Lidar: []
Step 6
Lidar: [[ 1.8227493   0.46455875 -0.01575875]
 [ 1.876159    0.4751287  -0.07096738]
 [ 1.8111563   0.45842275 -0.04207077]
 ...
 [ 1.5659422  -0.07916816  0.23125394]
 [ 1.5796202  -0.08043749  0.25589088]
 [ 1.595224   -0.08050884  0.28163904]]
Step 7
Lidar: [[ 1.8227493   0.46455875 -0.01575875]
 [ 1.876159    0.4751287  -0.07096738]
 [ 1.8111563   0.45842275 -0.04207077]
 ...
 [ 1.5659422  -0.07916816  0.23125394]
 [ 1.5796202  -0.08043749  0.25589088]
 [ 1.595224   -0.08050884  0.28163904]]
Step 8
Lidar: [[ 1.8227493   0.46455875 -0.01575875]
 [ 1.876159    0.4751287  -0.07096738]
 [ 1.8111563   0.45842275 -0.04207077]
 ...
 [ 1.5659422  -0.07916816  0.23125394]
 [ 1.5796202  -0.08043749  0.25589088]
 [ 1.595224   -0.08050884  0.28163904]]
Step 9
Lidar: [[ 1.8227493   0.46455875 -0.01575875]
 [ 1.876159    0.4751287  -0.07096738]
 [ 1.8111563   0.45842275 -0.04207077]
 ...
 [ 1.5659422  -0.07916816  0.23125394]
 [ 1.5796202  -0.08043749  0.25589088]
 [ 1.595224   -0.08050884  0.28163904]]
2025-08-05T15:49:23Z [15,953ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
[16.086s] Simulation App Shutting Down
```

Unfortunately I have to run headless, because I want to run simulations batchwise in a dockerized environment.

This is the script I am running:
```
from isaacsim import SimulationApp

config = {
    "headless": False
}
simulation_app = SimulationApp(config)

import omni.replicator.core as rep
from omni.replicator.core import AnnotatorRegistry

stage_path = "test40.usd"
simulation_app.context.open_stage(stage_path)
lidar_path = "/World/Example_Rotary"
lidar_rp = rep.create.render_product(
    lidar_path, (1, 1), "lidar")
lidar_anno = AnnotatorRegistry.get_annotator("IsaacExtractRTXSensorPointCloud")
lidar_anno.attach(lidar_rp)

for i in range(10):
    print(f"Step {i}")
    rep.orchestrator.step()
    lidar_data = lidar_anno.get_data()["data"]
    print(f"Lidar: {lidar_data}")

rep.orchestrator.wait_until_complete()
lidar_anno.detach()
lidar_rp.destroy()
simulation_app.close()
```

### Isaac Sim version

5.0.0 updated from main branch on August 5th 2025

### Operating System (OS)

Ubuntu 22

### GPU Name

A40

### GPU Driver and CUDA versions

550.127.05, 12.4

zwdoescode, 2025-08-12:
Hi @simplelife2010 Thanks for your repro files! I am able to reproduce the issue. Let me reach out to the internal team.

cloudending, 2025-08-21:
I have same problem. 
I use simulation_app.update instead of rep.orchestrator.step. and the lidar data is not empty. I can not tell why
what is the differences between simulation_app.update and rep.orchestrator.step.
BTW. rep.orchestrator.step can produce lidar point cloud at isaacsim 4.5.

CoderMoYao, 2025-09-02:
if  i want use" IsaacExtractRTXSensorPointCloud ",what should i do,the application  I develop use "IsaacExtractRTXSensorPointCloud" run ok,"IsaacCreateRTXLidarScanBuffer" point cloud  alway null.