isaac-sim/IsaacSim #111: Action Graph doesn't work when loaded as a Payload

cjenken-ume, 2025-08-05:
### Description

When loading a payload that has an Action Graph in it, it won't run.  The file will run just fine when using File -> Open but when I drag it into the stage to load it as a Payload it won't run unless I create another Action Graph in the stage.

### Steps to reproduce

Create an Action Graph that does something on playback tick.  Press play to make sure that it's working.
Save the file.
Open a new empty stage
Drag the newly created file with the Action Graph into it from the Content Browser and drop it into the stage (or use File -> Add Payload) 
Press play and notice it no longer works.
Add an Action Graph to the Stage 
Press Play and now it should work.

I've attached a simple file with a working action graph to be used for testing.

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Loading a payload with a working action graph plays without having to do anything other than pressing the play button.

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 22.04

### GPU Name

Nvidia A10G

### GPU Driver and CUDA versions

GPU Driver: 550.54.17 - CUDA: 12.4.1

### Additional information

Action Graph test file.  When you press play the cube should translate in space.  When you stop it should revert to it's starting position.

[ActionGraphMoveCube.zip](https://github.com/user-attachments/files/21607959/ActionGraphMoveCube.zip)

vick-yu, 2025-08-06:
I was unable to reproduce the bug. After following the reproduction steps, I observed that the cube did translate in space when pressing play, as it should. Could you confirm if you are on the latest repository update (RC.35)?

cjenken-ume, 2025-08-07:
Oh!  Thanks for that.  I was using an older RC (5.0.0-rc.23).  The new RC.35 fixes this bug.

closed 2025-08-07 by cjenken-ume