isaac-sim/IsaacSim #119: How to add mesh to 3DGS

superjuices, 2025-08-08:
### Description

After the ply rebuilt by 3dgs was converted to usd and imported into isaacsim, I found that there was no mesh, but I saw in the official example that there seemed to be

[image]
[image]
[image]

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 22

### GPU Name

RTX 4090 D

### GPU Driver and CUDA versions

CUDA12.8

MiaoMieu, 2025-08-11:
Same question, also, i can't find this OmniNuRecVolumeAPI , where did you find it after importing 3dgs-usdz?

saulgooodman, 2025-08-14:
SAME QUESTION!

vick-yu, 2025-08-20:
The mesh in our datasets was generated by the NuRec workflow, not a general 3DGS reconstruction. 

For manually adding a ground plane to a 3DGRUT scene, you can refer to the ["How to Instantly Render Real-World Scenes in Interactive Simulation"](https://developer.nvidia.com/blog/how-to-instantly-render-real-world-scenes-in-interactive-simulation/) blog.