isaac-sim/IsaacSim #139: Segment a Point Cloud Failed

LeZheng-x, 2025-08-14:
### Description

 I ran the **Segment a Point Cloud** tutorial on the official [website](https://docs.isaacsim.omniverse.nvidia.com/latest/sensors/isaacsim_sensors_physx_lidar.html#segment-a-point-cloud). 
But the expected results are inconsistent with the tutorial:
This is the expected output from the tutorial:
[image]
This is the result of my actual operation(Semantic ID is not output and the point cloud color is consistent):
[image]
This is the code copied from the official website:
[code.txt](https://github.com/user-attachments/files/21775141/code.txt)

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 20.04

### GPU Name

RTX 2080Ti

### GPU Driver and CUDA versions

CUDA 11.8

zwdoescode, 2025-08-19:
Thanks for reporting this issue @LeZheng-x. Let me reach out to the appropriate person to look into this.

zwdoescode, 2025-08-20:
@LeZheng-x I confirmed with the internal team that this example doesn't work anymore and we will fix the tutorial. Instead of using PhysX lidar, you can try with RTX lidar which has the segmentation capability.

LeZheng-x, 2025-08-21:
Thanks, I'll close with the comment.

mgrova, 2025-08-31:
Hello!! @zwdoescode , how i can do that with the RTX Lidar? I'm trying to figure out how to do it, but I can't, because when I manage to extract the segmentation IDs, they aren't reconstructed correctly or the cloud is incomplete.

Bests,
Marco.

zwdoescode, 2025-09-19:
@mgrova @LeZheng-x 
You should be able to use PhysX lidar to segment a point cloud. All you need to do is use `get_prim_data` instead of `get_semantic_data` in the code snippet.