isaac-sim/IsaacSim #140: Hydra Engine Error: "Invalid USD RenderProduct Prim" After URDF Import

LinFei83, 2025-08-15:
### Description

Hello Team,

I am encountering a critical rendering error in Isaac Sim after importing a URDF file. The viewport fails to render, and the console shows errors related to omni.hydra, preventing me from visualizing or interacting with the imported model.

Steps to Reproduce:

Launch Isaac Sim 5.0.0.
Create a new, empty stage.
Navigate to File > Import > URDF.
Select a valid URDF file.
Keep the default import settings and click the "Import" button.
Expected Behavior:

The robot model should be imported successfully and rendered correctly in the viewport without any critical errors.

Actual Behavior:

The import process seems to finish, but the viewport either goes black or freezes. The console logs display the following critical errors, indicating a failure within the Hydra rendering engine:

<TEXT>
2025-08-15T07:46:09Z [3,728,909ms] [Error] [omni.hydra] Invalid USD RenderProduct Prim: /Render/OmniverseKit/HydraTextures/omni_kit_widget_viewport_ViewportTexture_0
2025-08-15T07:46:09Z [3,728,909ms] [Error] [omni.usd] HydraEngine::render failed to end the compute graph: error code 6
This suggests that the render target for the viewport became invalid, causing the render pipeline to fail.

### Isaac Sim version

5.0.0

### Operating System (OS)

Ubuntu 24.04

### GPU Name

RTX 3090

### GPU Driver and CUDA versions

Driver Version: 575.64.03      CUDA Version: 12.9

### Logs

2025-08-15T07:46:09Z [3,728,909ms] [Error] [omni.hydra] Invalid USD RenderProduct Prim: /Render/OmniverseKit/HydraTextures/omni_kit_widget_viewport_ViewportTexture_0
2025-08-15T07:46:09Z [3,728,909ms] [Error] [omni.usd] HydraEngine::render failed to end the compute graph: error code 6

### Additional information

https://github.com/MuammerBay/isaac_so_arm101/blob/4faf0f61d1a7fad8c45fe9ec911d79991e44052c/URDF/SO_5DOF_ARM100_8j_URDF.SLDASM/urdf/SO_5DOF_ARM100_8j_URDF.SLDASM.urdf

LeoJhonSong, 2025-08-31:
same error. As I remember, I'm able to import URDF with  "Create in Stage" selected, but a while later (did not restart isaacsim),  when I try to import another URDF, I encountered this bug

vick-yu, 2025-10-06:
@LinFei83 @LinFei83 is this issue still present? I encountered multiple issues when attempting to import your .urdf file, which appear to be related but distinct from the original Hydra rendering error.

```
2025-10-06T06:56:28Z [386,232ms] [Warning] [isaacsim.asset.importer.urdf] The path SO_5DOF_ARM100_8j_URDF.SLDASM is not a valid usd path, modifying to SO_5DOF_ARM100_8j_URDF_SLDASM
2025-10-06T06:56:38Z [396,399ms] [Warning] [isaacsim.asset.importer.urdf] ROS_PACKAGE_PATH not defined, will skip checking ROS packages
2025-10-06T06:56:38Z [396,399ms] [Warning] [isaacsim.asset.importer.urdf] Path: package://SO_5DOF_ARM100_8j_URDF.SLDASM/meshes/Base.STL not found
2025-10-06T06:56:38Z [396,399ms] [Warning] [isaacsim.asset.importer.urdf] Failed to resolve mesh 'package://SO_5DOF_ARM100_8j_URDF.SLDASM/meshes/Base.STL'
2025-10-06T06:56:38Z [396,399ms] [Warning] [isaacsim.asset.importer.urdf] Prim /visuals/Base/Base not created                     
2025-10-06T06:56:38Z [396,404ms] [Warning] [isaacsim.asset.importer.urdf] ROS_PACKAGE_PATH not defined, will skip checking ROS packages
2025-10-06T06:56:38Z [396,404ms] [Warning] [isaacsim.asset.importer.urdf] Path: package://SO_5DOF_ARM100_8j_URDF.SLDASM/meshes/Base.STL not found
2025-10-06T06:56:38Z [396,404ms] [Warning] [isaacsim.asset.importer.urdf] Failed to resolve mesh 'package://SO_5DOF_ARM100_8j_URDF.SLDASM/meshes/Base.STL'
2025-10-06T06:56:38Z [396,404ms] [Warning] [isaacsim.asset.importer.urdf] Prim /colliders/Base/Base not created                                           
2025-10-06T06:56:38Z [396,406ms] [Error] [omni.kit.commands.command] Failed to execute a command: URDFImportRobot.
  File "/home/vyu/isaacsim-5.0/exts/isaacsim.asset.importer.urdf/isaacsim/asset/importer/urdf/impl/extension.py", line 426, in check_callback
    self.start_import(checked=True, path=path)                                                                                                                       
  File "/home/vyu/isaacsim-5.0/exts/isaacsim.asset.importer.urdf/isaacsim/asset/importer/urdf/impl/extension.py", line 472, in start_import               
    self._load_robot(dest_path, path)                                                                                                                                
  File "/home/vyu/isaacsim-5.0/exts/isaacsim.asset.importer.urdf/isaacsim/asset/importer/urdf/impl/extension.py", line 586, in _load_robot
    omni.kit.commands.execute(                                                                                                                                       
  File "/home/vyu/isaacsim-5.0/extscache/omni.kit.commands-1.4.10+8131b85d.lx64.r.cp311/omni/kit/commands/command.py", line 463, in execute
    result = omni.kit.undo.execute(command, name, kwargs)                                                                                                            
[...skipped...]               
  File "/home/vyu/isaacsim-5.0/extscache/omni.kit.commands-1.4.10+8131b85d.lx64.r.cp311/omni/kit/undo/undo.py", line 78, in execute        
    result = _execute(command, name, level, history_key) 
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vyu/isaacsim-5.0/extscache/omni.kit.commands-1.4.10+8131b85d.lx64.r.cp311/omni/kit/undo/undo.py", line 459, in _execute
    raise error                                         
  File "/home/vyu/isaacsim-5.0/extscache/omni.kit.commands-1.4.10+8131b85d.lx64.r.cp311/omni/kit/undo/undo.py", line 420, in _execute
    result = command.do()                                                                                                                                            
             ^^^^^^^^^^^^
  File "/home/vyu/isaacsim-5.0/exts/isaacsim.asset.importer.urdf/isaacsim/asset/importer/urdf/impl/commands.py", line 145, in do     
    return self._urdf_interface.import_robot(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                                                                                                                                                                     
 <class 'RuntimeError'> Used null prim       
2025-10-06T06:56:38Z [396,409ms] [Error] [omni.ui.python] RuntimeError: Accessed invalid null prim

At:                                   
  /home/vyu/isaacsim-5.0/exts/isaacsim.asset.importer.urdf/isaacsim/asset/importer/urdf/impl/extension.py(594): _load_robot
  /home/vyu/isaacsim-5.0/exts/isaacsim.asset.importer.urdf/isaacsim/asset/importer/urdf/impl/extension.py(472): start_import
  /home/vyu/isaacsim-5.0/exts/isaacsim.asset.importer.urdf/isaacsim/asset/importer/urdf/impl/extension.py(426): check_callback
```