isaac-sim/IsaacSim #152: scene_based_sdg_utils not found when running Python scripts

Jiarx3, 2025-08-19:
### Description

I tried to import scene_based_sdg_utils in a Python script, but it immediately failed with a ModuleNotFoundError. Soon after that, the simulator crashed. 
I assumed this package would be included with the simulator since some sample code references it, but it doesn’t seem to exist in the environment.

### Isaac Sim version

5.0.0

### Operating System (OS)

Windows 10

### GPU Name

RTX 4060 Laptop GPU

### GPU Driver and CUDA versions

Driver 577.00, CUDA 12.9

### Logs

```
Current thread 0x00007b90 (most recent call first):
  <no Python frame>
2025-08-19T08:21:23Z [0ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash] A crash has occurred.  If a debugger should be attached, please set the '/crashreporter/debuggerAttachTimeoutMs' setting to a timeout in milliseconds.  This can be used to allow the crash reporter to wait for up to that long for a debugger to attach before processing or sending the crash report.
2025-08-19T08:21:23Z [160ms] [Warning] [carb.crashreporter-breakpad.plugin] Launched process 'py-spy.exe' as PID 28912
2025-08-19T08:21:23Z [419ms] [Warning] [carb.crashreporter-breakpad.plugin] Launched process 'crashreport.gui.exe' as PID 30612
2025-08-19T08:21:23Z [443ms] [Error] [carb.crashreporter-breakpad.plugin] 'crashreport.gui.exe' exited with code 2

2025-08-19T08:21:23Z [462ms] [Error] [carb.crashreporter-breakpad.plugin] [crash] Wrote dump file 'c:/5.0/isaac-sim-standalone-5.0.0-windows-x86_64/kit/data/Kit/Isaac-Sim Python/5.0\818785d8-ac09-4c7e-9491-edb9658b9d1b.dmp'
2025-08-19T08:21:23Z [463ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash] preventing upload of minidump due to user opt-out: 'c:/5.0/isaac-sim-standalone-5.0.0-windows-x86_64/kit/data/Kit/Isaac-Sim Python/5.0\818785d8-ac09-4c7e-9491-edb9658b9d1b.dmp'
2025-08-19T08:21:23Z [463ms] [Error] [carb.crashreporter-breakpad.plugin] [crash]     dump file size is 8315988 bytes, file is readable.
2025-08-19T08:21:24Z [1,704ms] [Fatal] [carb.crashreporter-breakpad.plugin] [crash] Crash detected in pid 45760 thread 28176
2025-08-19T08:21:24Z [1,705ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash] Crash metadata:
2025-08-19T08:21:24Z [1,705ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  BuildGitlabJobID = '182268560'
2025-08-19T08:21:24Z [1,705ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  CarbSdkVersion = '206.4+release.9396.3c6f8191.gl'
2025-08-19T08:21:24Z [1,706ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  CrashTime = 'Tue Aug 19 08:21:23 2025 GMT'
2025-08-19T08:21:24Z [1,706ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  DriverShaderCacheWrapper = 'disabled'
2025-08-19T08:21:24Z [1,707ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  DumpId = '818785d8-ac09-4c7e-9491-edb9658b9d1b'
2025-08-19T08:21:24Z [1,707ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  LastUploadStatus = '0'
2025-08-19T08:21:24Z [1,707ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  ProductName = 'OmniverseKit'
2025-08-19T08:21:24Z [1,708ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  PythonTracebackStatus = ''py-spy.exe' successfully wrote info to c:/5.0/isaac-sim-standalone-5.0.0-windows-x86_64/kit/data/Kit/Isaac-Sim Python/5.0/818785d8-ac09-4c7e-9491-edb9658b9d1b.py.txt'
2025-08-19T08:21:24Z [1,708ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  RetryCount = '0'
2025-08-19T08:21:24Z [1,708ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  StartupTime = '1755591642'
2025-08-19T08:21:24Z [1,708ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  UploadSuccessful = '0'
2025-08-19T08:21:24Z [1,709ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  UptimeSeconds = '41'
2025-08-19T08:21:24Z [1,709ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  UserStory = ''
2025-08-19T08:21:24Z [1,709ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  UserStoryStatus = ''crashreport.gui.exe' exited with code 2'
2025-08-19T08:21:24Z [1,710ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  Version = '107.3.1+production.206797.8131b85d.gl'
2025-08-19T08:21:24Z [1,710ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  aftermath_enabled = '1'
2025-08-19T08:21:24Z [1,710ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  aftermath_flags = '3'
2025-08-19T08:21:24Z [1,711ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  aftermath_status = 'auto-enabled'
2025-08-19T08:21:24Z [1,711ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  aftermath_version = '2024.1'
2025-08-19T08:21:24Z [1,711ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  appName = 'Isaac-Sim Python'
2025-08-19T08:21:24Z [1,712ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  appState = 'started'
2025-08-19T08:21:24Z [1,712ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  appVersion = '5.0.0'
2025-08-19T08:21:24Z [1,712ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  autoloadExts = ''
2025-08-19T08:21:24Z [1,713ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  buildBranch = 'production'
2025-08-19T08:21:24Z [1,713ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  buildCi = 'gl'
2025-08-19T08:21:24Z [1,713ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  buildConfig = 'release'
2025-08-19T08:21:24Z [1,714ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  buildGitlabJobName = 'kit-build-release-windows-x86_64'
2025-08-19T08:21:24Z [1,714ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  buildGitlabJobStage = 'kit-build'
2025-08-19T08:21:24Z [1,715ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  buildGitlabPipelineID = '30713850'
2025-08-19T08:21:24Z [1,715ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  buildHash = '8131b85d'
2025-08-19T08:21:24Z [1,716ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  buildMajor = '107'
2025-08-19T08:21:24Z [1,716ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  buildMinor = '3'
2025-08-19T08:21:24Z [1,716ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  buildNumber = '206797'
2025-08-19T08:21:24Z [1,717ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  buildPatch = '1'
[... 346 lines cut ...]
2025-08-19T08:21:24Z [1,740ms] [Fatal] [carb.crashreporter-breakpad.plugin] [crash] Thread 28176 backtrace follows:
2025-08-19T08:21:24Z [1,740ms] [Fatal] [carb.crashreporter-breakpad.plugin] 000: ntdll.dll!RtlAcquireSRWLockShared+0x3a
2025-08-19T08:21:24Z [1,741ms] [Fatal] [carb.crashreporter-breakpad.plugin] 001: omni.fabric.plugin.dll!carbOnPluginStartup+0x1fb05
2025-08-19T08:21:24Z [1,741ms] [Fatal] [carb.crashreporter-breakpad.plugin] 002: omni.fabric.plugin.dll!carbOnPluginStartup+0x8d94b
2025-08-19T08:21:24Z [1,742ms] [Fatal] [carb.crashreporter-breakpad.plugin] 003: omni.graph.core.plugin.dll!omni::graph::core::debug::tokenAsString+0x74df9
2025-08-19T08:21:24Z [1,742ms] [Fatal] [carb.crashreporter-breakpad.plugin] 004: omni.graph.core.plugin.dll!omni::graph::core::debug::tokenAsString+0x77f9f
2025-08-19T08:21:24Z [1,742ms] [Fatal] [carb.crashreporter-breakpad.plugin] 005: omni.graph.core.plugin.dll!omni::graph::core::debug::tokenAsString+0x78271
2025-08-19T08:21:24Z [1,743ms] [Fatal] [carb.crashreporter-breakpad.plugin] 006: omni.graph.core.plugin.dll!carbOnPluginShutdown+0x12d
2025-08-19T08:21:24Z [1,743ms] [Fatal] [carb.crashreporter-breakpad.plugin] 007: carb.dll!carbReallocate+0xa68a
2025-08-19T08:21:24Z [1,743ms] [Fatal] [carb.crashreporter-breakpad.plugin] 008: carb.dll!releaseFrameworkAndTerminate+0x1d922
2025-08-19T08:21:24Z [1,744ms] [Fatal] [carb.crashreporter-breakpad.plugin] 009: carb.dll!omniReleaseStructuredLog+0x7018
2025-08-19T08:21:24Z [1,744ms] [Fatal] [carb.crashreporter-breakpad.plugin] 010: carb.dll!releaseFrameworkAndTerminate+0x1e5e9
2025-08-19T08:21:24Z [1,747ms] [Fatal] [carb.crashreporter-breakpad.plugin] 011: carb.scripting-python.plugin.dll!+0x1c366
2025-08-19T08:21:24Z [1,748ms] [Fatal] [carb.crashreporter-breakpad.plugin] 012: carb.scripting-python.plugin.dll!+0x135d
2025-08-19T08:21:24Z [1,748ms] [Fatal] [carb.crashreporter-breakpad.plugin] 013: carb.scripting-python.plugin.dll!+0xfd3b
2025-08-19T08:21:24Z [1,749ms] [Fatal] [carb.crashreporter-breakpad.plugin] 014: python311.dll!PyCFunction_GetFlags+0xe2c
2025-08-19T08:21:24Z [1,749ms] [Fatal] [carb.crashreporter-breakpad.plugin] 015: python311.dll!PyObject_Call+0xa8
[... cut ...]
2025-08-19T08:21:24Z [1,750ms] [Fatal] [carb.crashreporter-breakpad.plugin] 018: python311.dll!Py_RunMain+0x41
2025-08-19T08:21:24Z [1,750ms] [Fatal] [carb.crashreporter-breakpad.plugin] 019: python311.dll!Py_Main+0x26
2025-08-19T08:21:24Z [1,754ms] [Fatal] [carb.crashreporter-breakpad.plugin] 020: python.exe!+0x123c
2025-08-19T08:21:24Z [1,754ms] [Fatal] [carb.crashreporter-breakpad.plugin] 021: KERNEL32.DLL!BaseThreadInitThunk+0x17
2025-08-19T08:21:24Z [1,754ms] [Fatal] [carb.crashreporter-breakpad.plugin] 022: ntdll.dll!RtlUserThreadStart+0x2c
```

vick-yu, 2025-08-19:
As a quick check, have you tried running the built-in example that uses this module? The script located at `standalone_examples/replicator/scene_based_sdg/scene_based_sdg.py` ran successfully on my machine.

This could help verify if the issue is related to the environment setup or the specific implementation in your script. Let me know if that example works for you!

vick-yu, 2025-08-22:
Please start with the code that works and slowly add your own logic to create a minimal reproducible script.

closed 2025-09-21 by vick-yu