isaac-sim/IsaacSim #208: IsaacSim crashes on importing OpenArm URDF

SakshayMahna, 2025-09-20:
### Description

While trying to import OpenArm URDF, IsaacSim crashes with this error. Attaching the folder and error message here.

[image]

This is the repository: https://github.com/SakshayMahna/openarm_description

Steps to reproduce:

1. Clone the above repository
2. Start IsaacSim
3. Import URDF, select `openarm_bimanual.urdf` from the `openarm_description` folder.
4. IsaacSim crashes

### Isaac Sim version

5.0.0

### Operating System (OS)

AWS Workstation

### GPU Name

AWS Workstation

### GPU Driver and CUDA versions

AWS Workstation

freckletonj, 2025-10-03:
This is so hard to debug, because every change to my URDF results in a segfault. It'd be really nice if we could validate URDFs safely.

For instance, in trying to make a minimal repro, I found that the most minimal thing segfaults:

```xml
<?xml version="1.0" ?>
<robot name="igus_rebel_6dof">
</robot>
```

I think it segfaults for other reasons too, but I'm struggling to enumerate the reasons.

Can mesh geometry be `.dae`? I've seen some imports work using `.stl`, and another segfaults if `.dae` is mentioned for mesh geom.

freckletonj, 2025-10-03:
Ok. I'm on day 3 of debugging how to import a valid URDF. It appears `.dae` is not supported for mesh geom.

This is extra frustrating since Isaac Sim *can* (ostensibly) import `.dae`.

**PLEASE** #1 write a check for the existence of `.dae` for mesh geom in URDF and give a more sensible feedback. It took me 2.5 days to arrive at this awareness, and I'm sure I'm not the only one.

**PLEASE** #2 Isaac Sim should never segfault. If you want to flush out dozens of bugs, simply try to import this file: https://github.com/CommonplaceRobotics/iRC_ROS/blob/humble/irc_ros_description/urdf/igus_rebel_6dof.urdf.xacro . You can use ros or xacrodoc to get a URDF from it. I don't have ros, so I used `xacrodoc`.

So I've gone through and converted all `.dae -> .stl` using `blender`. 

This can ONLY be imported as "Referenced Model", not as "Create in Stage". If I do create in stage, the viewport freezes up and I get 100 errors per second in the console (none of which indicate what's breaking).

**PLEASE** #3 find what bugs are disallowing "Create in Stage".

Now it imports without a segfault, but all my joints are flipped around:

[image]

We all have expensive hardware from you. Please dont leave us hanging.

freckletonj, 2025-10-03:
Just documenting this here since the maintainers are unresponsive (been 2 weeks since OP).

`dae -> stl` seems necessary for avoiding segfaults.

You can convert `dae -> stl` using blender, but this results in the above origin shifts:

```sh
blender --background --python-expr "import bpy; bpy.ops.wm.collada_import(filepath='Joint0Coll.dae'); bpy.ops.export_mesh.stl(filepath='Joint0Coll.stl')"
```

You can convert `dae -> stl` using FreeCAD, but now you get new different origin funkery, and in simulation, the collision meshes do not stay aligned with the visual meshes:

```sh
freecad -c "import FreeCAD, Mesh; from importers import importDAE; importDAE.open('Joint0Coll.dae'); Mesh.export(FreeCAD.activeDocument().Objects, 'Joint0Coll.stl')"
```

[image]

freckletonj, 2025-10-03:
This has been a ride.

So typical URDF stls/daes are in mm, and IsaacSim expects m.

You have to:

1. convert all dae to stl, making sure to scale to meters (so, mm*1/1000)
2. move all mesh files into a folder next to the urdf (ie can't be a directory backword in the tree)
3. remove `package://` from the asset name, bc I think IsaacSim doesn't play well with that
4. import only as "Referenced", not "Create in Stage" (or it'll break IsaacSim and throw errors)
5. manually trim out XML, such as ros2 tags in my case

[image]

For my case, here's a convenient bash script using freecad. It:

1. bakes translation information into the mesh vertex (instead of treating translation as a parameter that will be dropped when converting to stl)
2. scale by `0.001`, and output stl
3. then by hand you still need to move all these new stl files

Note: the imported geom still doesn't respond to `Shift-click` forces during play, so something's still not right.

```sh
for f in Joint0Coll.dae Joint0.dae Joint1Coll.dae Joint1.dae Joint2Coll.dae Joint2.dae Joint3Coll.dae Joint3.dae Joint4Coll.dae Joint4.dae Joint5Coll.dae Joint5.dae Joint6Coll.dae Joint6.dae; do
  freecad -c "import FreeCAD, Mesh
from importers import importDAE
importDAE.open('$f')
doc=FreeCAD.activeDocument()
doc.recompute()
meshes=[]
for obj in doc.Objects:
    m=obj.Mesh.copy()
    m.transform(obj.Placement.toMatrix())
    s=FreeCAD.Matrix()
    s.A11=s.A22=s.A33=0.001
    m.transform(s)
    obj.Mesh = m
    print(f'{m=}')
doc.recompute()
Mesh.export(FreeCAD.activeDocument().Objects, '${f%.dae}.stl')"
done
```

All the previously mentioned comments/issues still apply. IsaacSim is flaky, segfaults, breaks with `.dae`, throws a bajillion errors, "no prim" errors without explanation, etc. This is just a laborious workaround and should not close this ticket.

freckletonj, 2025-10-04:
shoot, more weirdness, it successfully imports the STLs, but duplicates them, once with an origin at the correct location, and once near the fixed joint location. The STLs are fine btw, no duplicates. And the URDF only invokes these meshes once. It's IsaacSim being weird again.

The following show a single joint highlighted at a time, but theyve been duplicated.

[image]

[image]

[image]