isaac-sim/IsaacSim #226: Pointcloud output test failed with a crash

zhxchen-se, 2025-09-29:
### Description

While running the `test_pointcloud_output` test, the validation did not pass.  The `Camera.get_pointcloud` returns a point cloud with fewer points than expected (e.g., (59536, 3) instead of the expected (65536, 3) for a 256×256 camera). The script ends with a crash finally.

```
def test_pointcloud_output(
    camera_default: Camera, camera_cpu: Camera, camera_cuda: Camera, pointcloud_shape: tuple
) -> bool:
    """Tests the Pointcloud output for cameras with different annotator devices."""
    print("=" * 80)
    print("Testing: pointcloud")

    world_frame_list = [True, False]

    print("-" * 40)
    print("Get pointcloud data using Camera(annotator_device=None):")
    print("---")
    test_1_passed = True
    for world_frame in world_frame_list:
        print(f"world_frame: {world_frame}")
        print("---")
        default_pointcloud = camera_default.get_pointcloud(world_frame=world_frame)
        default_pointcloud_cpu = camera_default.get_pointcloud(device="cpu", world_frame=world_frame)
        default_pointcloud_cuda = camera_default.get_pointcloud(device="cuda", world_frame=world_frame)
        success = test_camera_annotator_data(
            test_name=f"get_pointcloud_device_None_world_frame_{world_frame}",
            data=default_pointcloud,
            expected_class=np.ndarray,
            expected_dtype=np.float32,
            expected_shape=pointcloud_shape,
        )
        test_1_passed = test_1_passed and success
        success = test_camera_annotator_data(
            test_name=f"get_pointcloud_device_cpu_world_frame_{world_frame}",
            data=default_pointcloud_cpu,
            expected_class=np.ndarray,
            expected_dtype=np.float32,
            expected_shape=pointcloud_shape,
        )
        test_1_passed = test_1_passed and success
        success = test_camera_annotator_data(
            test_name=f"get_pointcloud_device_cuda_world_frame_{world_frame}",
            data=default_pointcloud_cuda,
            expected_class=wp.array,
            expected_dtype=wp.types.float32,
            expected_shape=pointcloud_shape,
        )
        test_1_passed = test_1_passed and success
        print("---")
    if test_1_passed:
        print("[PASS]")
    else:
        print("[FAIL]")
    print("---")

    print("-" * 40)
    print("Get pointcloud data using Camera(annotator_device='cpu'):")
    print("---")
    test_2_passed = True
    for world_frame in world_frame_list:
        print(f"world_frame: {world_frame}")
        print("---")
        cpu_pointcloud = camera_cpu.get_pointcloud(world_frame=world_frame)
        cpu_pointcloud_cpu = camera_cpu.get_pointcloud(device="cpu", world_frame=world_frame)
        cpu_pointcloud_cuda = camera_cpu.get_pointcloud(device="cuda", world_frame=world_frame)
        success = test_camera_annotator_data(
            test_name=f"get_pointcloud_device_None_world_frame_{world_frame}",
            data=cpu_pointcloud,
            expected_class=np.ndarray,
            expected_dtype=np.float32,
[... 132 lines cut ...]
2025-09-29T13:24:50Z [55,877ms] [Error] [__main__] Data shape is (59536, 3) but expected (65536, 3)
get_pointcloud_device_cuda_world_frame_False: data.shape: (59536, 3); dtype: <class 'warp.types.float32'>; type: <class 'warp.types.array'>
2025-09-29T13:24:50Z [55,878ms] [Error] [__main__] Data shape is (59536, 3) but expected (65536, 3)
---
[FAIL]
---
```

```
Current thread 0x00008938 (most recent call first):
  <no Python frame>
2025-09-29T13:25:17Z [0ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash] A crash has occurred.  If a debugger should be attached, please set the '/crashreporter/debuggerAttachTimeoutMs' setting to a timeout in milliseconds.  This can be used to allow 
the crash reporter to wait for up to that long for a debugger to attach before processing or sending the crash report.
2025-09-29T13:25:17Z [25ms] [Warning] [carb.crashreporter-breakpad.plugin] Launched process 'py-spy.exe' as PID 33532
2025-09-29T13:25:17Z [119ms] [Warning] [carb.crashreporter-breakpad.plugin] Launched process 'crashreport.gui.exe' as PID 30116
2025-09-29T13:25:17Z [140ms] [Error] [carb.crashreporter-breakpad.plugin] 'crashreport.gui.exe' exited with code 2

2025-09-29T13:25:17Z [150ms] [Error] [carb.crashreporter-breakpad.plugin] [crash] Wrote dump file 'c:/isaac-sim/kit/data/Kit/Isaac-Sim Python/5.0\153252ad-cae7-4263-abe7-a5ae17d54bf6.dmp'
2025-09-29T13:25:17Z [151ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash] preventing upload of minidump due to user opt-out: 'c:/isaac-sim/kit/data/Kit/Isaac-Sim Python/5.0\153252ad-cae7-4263-abe7-a5ae17d54bf6.dmp'
2025-09-29T13:25:17Z [153ms] [Error] [carb.crashreporter-breakpad.plugin] [crash]     dump file size is 8117660 bytes, file is readable.
2025-09-29T13:25:19Z [1,493ms] [Fatal] [carb.crashreporter-breakpad.plugin] [crash] Crash detected in pid 25064 thread 36148
2025-09-29T13:25:19Z [1,495ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash] Crash metadata:
2025-09-29T13:25:19Z [1,496ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  BuildGitlabJobID = '182268560'
2025-09-29T13:25:19Z [1,498ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash]  CarbSdkVersion = '206.4+release.9396.3c6f8191.gl'
```

zhxchen-se, 2025-10-04:
> [@zhxchen-se](https://github.com/zhxchen-se) could you please try the 5.1.0 OSS version? I don't see any issues/crash in 5.1.0 OSS

Is there a release version? Do I have to clone the repo and build from source?

zhxchen-se, 2025-10-04:
> [@zhxchen-se](https://github.com/zhxchen-se) could you please try the 5.1.0 OSS version? I don't see any issues/crash in 5.1.0 OSS
I cloned the latest GitHub repository and installed version 5.1.0 from source. After running the same script, I found that the issue no longer occurs. Maybe it's a legacy bug in version 5.0.0.

zwdoescode, 2025-10-05:
The stable version will be released this month. Stay tuned! I will close this ticket.

closed 2025-10-05 by zwdoescode